goecharger/go-eCharger-API-v2 #1: Versuch mit "filter" klappt nicht.

chk2902, 2021-09-12:
Direkt aus eurem Beispiel in der Doku:

`http://192.168.2.59/api/status?filter=[%22rfb%22,%22alw%22,%22acu%22,%22adi%22,%22amp%22]`

->

`{}`

`http://192.168.2.59/api/status?filter=[%22rfb%22,%22alw%22,%22acu%22,%22adi%22,%22amp%22]`

->

`{...Massen an Daten...}`

Ich hatte irgendwie mehr erwartet... "fwv" ist "051.4"

Was will die Box denn von mir?

Sillium007, 2021-09-13:
Kann ich bestätigen. Wenn ich einen Filter anwende bekomme ich auch nur eine leere Antwort

jenskaschub, 2021-09-13:
Wurde geändert, bitte versucht http://192.168.1.211/api/status?filter=acu,amp

Sillium007, 2021-09-13:
> Wurde geändert, bitte versucht http://192.168.1.211/api/status?filter=acu,amp

Funktioniert :)

chk2902, 2021-09-13:
Aber nicht in der Doku:

```
Danach können vereinzelte API keys abgefragt werden:

http://192.168.0.75/api/status?filter=["rfb","alw","acu","adi","amp"]
```

Sillium007, 2021-09-13:
Hab nen PR dafür gestellt. Mal sehen ob sie es selber fixen oder den erstmal übernehmen #2

peterpoetzi, 2021-09-13:
Seit Firmware 051.4 ist Filter jetzt kein JSON Array mehr, sondern nur eine Liste von Komma-separierten Strings

chk2902, 2021-09-13:
Heißt, man muss wieder die Firmware-Version zur Entscheidung nutzen, wie man die Daten übergibt...

**Bitte (!) macht das in der Doku klar**. Jede Änderung muss dokumentiert werden, ab welcher Firmware-Version ws gültig ist, und wie das vorher war.

sebnaf, 2021-09-13:
Idealerweise sind diese Parameter einmal fixiert und werden nicht mehr geändert! Alles andere ist wirklich unprofessionell.

Sillium007, 2021-09-13:
Idealerweise fragt man freundlich und macht kein Bashing wenn man etwas will. Ich denke nicht dass Go-E verpflichtet ist euch überhaupt eine Doku zu geben oder dass ihr einen Vertrag mit ihnen habt der alles festschreibt wie die Doku und API aussehen soll.
Ich bin froh dass Go-E recht offen mit den Dingen ist und die Infos zur Verfügung stellt und hoffe das Go-E durch das von eurer Seite aus unprofessionelle Verhalten sich nicht davon abbringen lässt weiterhin Informationen bereitzustellen.

sebnaf, 2021-09-13:
Es geht nicht um die Doku. Es geht um das Ändern von einmal definierten und kommunizierten Werten oder Verhalten.
go-e hat durchaus Interesse, deren Charger für diverse Integrationen zu verkaufen. Dies ist ein eindeutiger Vorteil, welcher auch auf der Webseite und im Shop hervorgehoben wird.
In diesem Fall wünsche ich mir - für go-e! - eine gewisse Stetigkeit der APIs, da sonst mittelfristig keiner Interesse mehr hat, mit den APIs zu arbeiten. Wenn dann ein anderer mit ähnlicher Box, ähnlichem Preis und funktionierenden APIs kommt, hätte ich als go-e Sorgen um den zukünftigen Marktanteil.

Kurzum: ich finde das Produkt toll, sehe aber in einigen Bereichen noch Luft nach oben.

Sillium007, 2021-09-13:
Natürlich ist das Ziel bestimmt es stabil zu halten, aber ich fand die Reaktionen etwas überzogen. Die Doku gibt es seit einer Woche und ist nicht vollständig derzeit, daher würde ich jetzt auch nicht das Problem so groß sehen dass es schon zig Firmen/Leute gibt die fertige Produkte haben die darauf basieren :)

Gut.. keiner von uns hat Einsicht in Go-E, aber ich denke durchaus dass sie die Ziele haben die Du beschreibst.

sebnaf, 2021-09-13:
Naja, auf genau diese Doku warten Entwickler seit Monaten. Die freuen sich, beginnen damit zu arbeiten und stellen nach nach knappen Woche fest: ach, hat sich was geändert.

peterpoetzi, 2021-09-13:
Hi, 

es tut uns leid dass sich der Filter geändert hat. Bei Tests hat sich herausgestellt dass das Parsen eines JSON Arrays sehr unperformant ist und daher nicht praktikabel. Wir werden aber die Doku dahingehend anpassen welche Firmware Version erforderlich ist.

sebnaf, 2021-09-13:
Hi @peterpoetzi Fehler und Verbesserungen passieren und sollten sein. Bitte einfach zeitnah und entsprechend dokumentieren - und alle sind glücklich, danke.

Sillium007, 2021-09-13:
Doku ist ja schon aktualisiert :)

chk2902, 2021-09-13:
@Sillium007 die API ist Bestandteil des Angebots: "Mit dem go-eCharger erhalten Sie ein System mit einer offenen HTTP API Schnittstelle (MQTT / REST). Dies ist vor allem dann interessant, wenn Sie eigene Applikationen anbinden (lassen) möchten. Beispielsweise eine Photovoltaik-Anlage. Durch die offene Schnittstelle ist sichergestellt, dass andere Hersteller Ihre Geräte mit dem go-eCharger kombinieren können. Die Dokumentation der HTTP API von go-e finden Sie hier"

closed 2021-09-13 by peterpoetzi