goecharger/go-eCharger-API-v2 #5: Wie kann man die Ladebereitschaft der Box setzen?

chk2902, 2021-09-14:
Wie kann man einstellen, dass die Box ladebereit ist (d.h. automatisch anfängt zu laden mit einem bestimmten Ladestrom, sobald ein Auto verbunden wird), oder dass die Box nicht anfängt zu laden, auch wenn ein Auto angesteckt wird?

Früher ging das mit "alw=1" oder "alw=0".

Nun ist "alw" readonly.

Sillium007, 2021-09-14:
http://*.*.*.*/api/set?frc=0
https://github.com/goecharger/go-eCharger-API-v1/issues/66

chk2902, 2021-09-14:
Danke - fragt sich, was "normal" ist (siehe Doku).

Das geht mir zu langsam... Wie @sebnaf schon mal geschrieben hatte (https://github.com/goecharger/go-eCharger-API-v1/issues/57), dauert es etliche Sekunden, bis "alw" auf false oder true ist.

OK, dann fällt "alw" raus und "frc" kommt rein. Weiß nicht, warum das nötig war. Gut, das hat 3 Zustände - aber für eine Steuerung braucht man nur 2 (die Funktion von "normal" ist unbekannt).

sebnaf, 2021-09-14:
@chk2902 Mit aktueller Firmware sollte dieses Delay behoben sein.

chk2902, 2021-09-14:
@sebnaf Hm, ich habe auf der Box die 051.4, und da tritt es noch auf. Wenn ich mich auf "alw" verlasse, muss man ziemlich lange warten, bis der Zustand dem "frc"-Wunsch entspricht.

Schlussendlich sieht es im Moment so aus:

```
    public boolean isPowerSwitchedOn()
        {
        if (getAPIVersion() == 2)
            {
            // "alw" braucht ein paar Sekunden nach dem Einschalten, man kann dem Wert nicht trauen...
            /*
            if (m_Obj.optBoolean("alw", false))
                {
                return (true);
                }
             */
            // der direkteste Weg
            switch (m_Obj.optInt("frc", 0))
                {
                default:
                    break;
                case 2:
                    return (true);
                case 0:
                    return (false);
                }
            // das Auto kann auch noch Hilfestellung geben
            switch (m_Obj.optInt("car", 0))
                {
                default:
                    break;
                case 2:
                case 3:
                    return (true);
                }
            return (false);
            }
        // alte Version:
        return (m_Obj.optInt("alw", 0) != 0);
        }
```

Sillium007, 2021-09-14:
Ich hab es nicht ausprobiert, aber brauchst Du die Rückmeldung über alw überhaupt. Beim setzen bekommst doch gleich zurück ob das setzen von frc geklappt hat. Würde das nicht als Status für Dich reichen?
Ich weiß ja nicht wie die weitere Logik dahinter aussieht. Nur mal als Idee

sebnaf, 2021-09-14:
Ich hätte ein wenig Bauchschmerzen damit, api v1 und v2 zu vermischen. Idealerweise habe ich auf dem go-e nur eine Version aktiv.

Sillium007, 2021-09-14:
Wie kommst du drauf dass es gemischt ist?

chk2902, 2021-09-14:
Meine App nutzt nur die Zustände der Antworten, das ist viel zuverlässiger - sonst würde es bedeuten, dass ich (wie lange???) den "alw"-Zustand ignorieren muss und mich ganz auf meinen "Gemerkt"-Zusatand verlassen muss - und wenn jemand dann mit einer anderen App etwas einstellt, bekommt dann meine App das nicht mehr mit wegen des eigenen, dann falschen Werts.

Ich finde, das ist Sache der Wallbox: ich setze "alw=1" (hat "früher" (TM) auch so funktioniert in der V1) oder nun "frc=2" - aber "alw" wechselt eben erst viel zu spät für die zuverlässige Nutzung als "Zustand der Box" - zumindest, wenn man im UI dann den "Start"- oder "Stop"-Button entsprechend en- oder disabled. Mag intern schon Sinn ergeben (das Auto lädt bei einem Kunden erst 10-40 Sekunden nach "alw=1" an der Hardware V3!), aber für die UI taugt das nicht.

sebnaf, 2021-09-14:
Ich glaube in dem Fall solltest Du ein Issue hier erstellen.

chk2902, 2021-09-14:
https://github.com/goecharger/go-eCharger-API-v2/issues/6

0xFEEDC0DE64, 2021-09-14:
https://github.com/goecharger/go-eCharger-API-v2/issues/6#issuecomment-918931536

closed 2021-09-17 by peterpoetzi