goecharger/go-eCharger-API-v2 #6: Wie kann man die Ladebereitschaft der Box erfahren (für die UI)

chk2902, 2021-09-14:
Siehe https://github.com/goecharger/go-eCharger-API-v2/issues/5

Problem ist, der Box zu übergeben, dass man laden will, z.B. wenn ein Benutzer auf einen Button "Start" drückt.

"alw" gab in V1 diesen Zustand sofort zurück nach dem Setzen. Passte.

"alw" in V2 aber erst nach etlichen Sekunden, wenn man "frc" auf 2 (oder 1) setzt.

Wenn man das Enablen und Disablen der Buttons (Start enabled, Stop disabled, wenn WB nicht ladebereit, und andersrum) nun an einem Zustand "festmachen" will, wie macht man das dann am Besten?

"frc" ist schon OK, aber was, wenn "frc" auf "0" steht? Was ist das für ein Zustand? An, Aus, Schrödingers Katze? Was kann man dann nutzen?

(Grund ist, dass ja mehrere Clients auf die Box zugreifen könnten, der Zustand der App also aus dem Status bestimmt werden muss und nicht aus der eigenen "ich weiß, der Benutzer hat Start gedrückt" Historie).

0xFEEDC0DE64, 2021-09-14:
Hi, ich hoffe ich kann hier Klarheit schaffen. Gedacht war, dass man über die API die gespeicherten Einstellungen setzt, und einmal in der Sekunde wird die tausende-Zeilen-lange Ladelogik mit genau diesen Einstellungen evaluiert. Teil des API-Status ist auch das letzte Ergebnis der API. Das letzte Ergebnis gilt mind. 1 Sekunde lang bis zum nächsten Evaluieren der Ladelogik.

"alw" zeigt hier als boolean, ob überhaupt Laden erlaubt ist.
"frc" definiert einen force-Zustand, der Neutral, ForceOff oder ForceOn sein kann. Bei neutral hängt die Ladeerlaubnis von tausenden Dingen ab, zB Energielimit erreicht, Scheduler, Awattar Preis, etc.

Mit ForceOn / ForceOff kann man hier sofort "einschreiten" und das gewünschte Ergebnis erzwingen. Es hat sich gezeigt, dass es oft zum Schluss vergessen wird und daher wird der ForceState beim Abstecken des Autos wieder auf Neutral zurück gestellt.

TLDR;
willst du sofort laden, setzte frc=2,
willst du, dass er sofort aufhört zu laden, setze frc=1
willst du wieder das ursprüngliche Verhalten mit allen Komfortfunktionen, setze frc=0

chk2902, 2021-09-14:
Danke.

Allerdings folgt das "alw" nicht dem "frc", sonst müßte ja beim nächsten Evaluieren das "alw" den zu erwartenden Wert haben. Dachte ich zumindest. OK, da spielt noch mehr rein als nur "frc".

Frage war: woran kann ich erkennen, ob "Start" oder "Stop" sinnvolle Kommandos sind (um die in der Oberfläche zu enablen)?

Frage 2 wäre dann: wie ist das mit "von außen gesteuerter" Ladung, z.B. PV-Überschuss: wenn "dws >= dwo" (Lademengenabbruch), was passiert dann bei "frc=2", wenn die Sonne so wunderschön scheint? Bleibt die Ladung gestoppt, oder lädt die WB weiter? "force" klingt ja nach "force", aber vermutlich und hoffentlich wird das dann durch die "Ladelogik" im Keim erstickt, wenn die Lademenge geladen ist. So wie früher auch.

chk2902, 2021-09-14:
D.h. es kann sein, dass "frc=2", aber Ladung findet trotzdem nicht statt? Woran erkenne ich das dann?

chk2902, 2021-09-14:
Also meine Aufgabe: ich möchte eine Ladung starten mit 16 A und 10 kWh Ladebeschränkung. Wie mache ich das über die App?

"amp=16&dwo=10000&frc=2" wäre mein Ansatz. Aber klappt dann die Ladeabschaltung auch?

0xFEEDC0DE64, 2021-09-16:
Ladeabschaltung bei frc=2 (ForceOn) wird nicht mehr durch geführt, es wird ja ForceOn gefordert.

Stell die Box im logic mode Default (lmo=3), frc auf Neutral oder Off und amp und dwo wie aus deinem beispiel, sollte dann klappen.

Hier bei mir das Example, ich wechsle von ForceOff auf Neutral, und auf einmal ist auch das laden erlaubt (wobei mein Fahrzeug leider schon voll war und deswegen aus dem car state Complete nicht mehr zurück auf Charging geht)
[image]

chk2902, 2021-09-17:
Danke, klappt so weit (dann fehlt nur noch die etwas ausführlichere Doku zu dem "lmo" ;))

Also Einschalten mit "`amp=16&dwo=10000&lmo=3&frc=0`", Beenden mit "`frc=1`"? Oder ist Beenden besser mit "`amp=0&frc=0`"?

chk2902, 2021-09-17:
OK, keine gute Idee...

"Server-Fehler: {"amp":"amp: failed: Value 0 exceeds range 6 to 32 (0)"} (com.android.volley.ServerError)"

peterpoetzi, 2021-09-17:
Wenn man möchte dass das Laden permanent nicht möglich ist, muss man Access Control auf Wait setzen. Der Force State wird auf Neutral zurückgesetzt wenn man das Auto absteckt. Die Ampere kann man nicht auf 0 setzen.

closed 2021-09-17 by peterpoetzi