goecharger/go-eCharger-API-v2 #7: API-Key "lot" fehlt in der V2-Dokumentation?

Ventusfahrer, 2021-09-16:
die restlichen Parameter für das Gruppen-Management sind noch Read/Write vorhanden.

Wenn ich mir die Parameterliste mit http://boxname/api/status hole, fehlt "lot" in der Liste.
Wenn ich mit http://boxname/api/set?lot=12 wird das Kommando akzeptiert, "lot" taucht danach als Parameter im Status auf. Allerdings gibt's vermutlich einen bösen Nebeneffekt: Beim Versuch sich die Werte in der App nochmal anzuschauen, verlangt die App ein neues Kennwort für den Zugriff auf die Box. Irgendwas hat's da zerschossen?

Grüße

Peter

0xFEEDC0DE64, 2021-09-17:
Hoffe mit meinem letzten Update der Api Keys wird das etwas klarer.

Den Effekt mit dem Passwort kann ich mir nicht erklären, sicher dass es mit dem api key lot zusammen hängt?

Ventusfahrer, 2021-09-17:
> Den Effekt mit dem Passwort kann ich mir nicht erklären, sicher dass es mit dem api key lot zusammen hängt?

so ganz genau kann ich nicht mehr nachvollziehen, was letztlich das Kennwort überschrieb.
Status war: wir haben 2 Home+ Ladeboxen, eine Box musste wegen massiver Instabilität (fw 51.4) auf die Werkseinstellungen zurückgesetzt werden. Danach wurde nach Aktivierung der Protokolle API-V1 und API-V2 auf der reset'eten Box für beide Boxen der api/status abgefragt. "lot" war nicht in der Statusliste aufgeführt. Nächster Schritt war das Setzen von "loe" -> true und das Setzen von "log" auf beiden, alles per API-V2. "lot" war zu diesem Zeitpunkt immer noch nicht im V2-Status in der Parameterliste beider Boxen zu finden. Nach dem Setzten von "lot" auf der gerade zurückgesetzten Box kam es zu dem beschriebenen Problem, das nur durch Neusetzen des Kennworts im Hotspot-Mode gefixt werden konnte. Nach dem Neustart dieser Box ging der App-Zugriff wieder wie erwartet. Jetzt wurde auf der 2. Box "lot" gesetzt, ohne dass das Kennwort-Problem auf der 2. Box auftrat. Auch auf der 2. Box tauchte "lot" in der Liste erst nach dem Setzen (..../api/set?lot=xx) auf.

Beide Boxen waren vor dem Reset bereits in einer Lastgruppe definiert deren Schlüssel uns durch den Reset verloren ging (wurde leider nicht außerhalb der App notiert)

JochenFriedrich, 2021-10-08:
Das gleiche Verhalten habe ich auch. Werde auch mal einen Factory Reset versuchen.

Mustafatarakci, 2022-04-12:
> Hoffe mit meinem letzten Api Anahtarlarını Güncelle, daha fazla bilgi.
> 
> Den Effekt mit dem Passwort kann ich mir nicht erklären, sicher dass es mit dem api anahtar partisi zusammen hängt?

Mustafatarakci, 2022-04-12:
<?xml version="1.0" encoding="UTF-8"?>

<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 https://maven.apache.org/xsd/maven-4.0.0.xsd">
  <modelVersion>4.0.0</modelVersion>
  <parent>
    <groupId>org.apache.accumulo</groupId>
    <artifactId>accumulo-project</artifactId>
    <version>2.0.1</version>
  </parent>
  <artifactId>accumulo-core</artifactId>
  <name>Apache Accumulo Core</name>
  <description>Apache Accumulo core libraries.</description>
  <dependencies>
    <dependency>
      <groupId>com.beust</groupId>
      <artifactId>jcommander</artifactId>
    </dependency>
    <dependency>
      <groupId>com.github.ben-manes.caffeine</groupId>
      <artifactId>caffeine</artifactId>
    </dependency>
    <dependency>
      <groupId>com.google.auto.service</groupId>
      <artifactId>auto-service</artifactId>
      <optional>true</optional>
    </dependency>
    <dependency>
      <groupId>com.google.code.gson</groupId>
      <artifactId>gson</artifactId>
    </dependency>
    <dependency>
      <groupId>com.google.guava</groupId>
      <artifactId>guava</artifactId>
    </dependency>
    <dependency>
      <groupId>com.google.protobuf</groupId>
      <artifactId>protobuf-java</artifactId>
    </dependency>
    <dependency>
      <groupId>commons-io</groupId>
      <artifactId>commons-io</artifactId>
    </dependency>
    <dependency>
      <groupId>commons-logging</groupId>
      <artifactId>commons-logging</artifactId>
    </dependency>
    <dependency>
      <groupId>jline</groupId>
      <artifactId>jline</artifactId>
    </dependency>
    <dependency>
      <groupId>log4j</groupId>
      <artifactId>log4j</artifactId>
    </dependency>
    <dependency>
      <groupId>org.apache.accumulo</groupId>
      <artifactId>accumulo-start</artifactId>
    </dependency>
    <dependency>
      <groupId>org.apache.commons</groupId>
      <artifactId>commons-collections4</artifactId>
    </dependency>
    <dependency>
      <groupId>org.apache.commons</groupId>
      <artifactId>commons-configuration2</artifactId>
    </dependency>
    <dependency>
      <groupId>org.apache.commons</groupId>
      <artifactId>commons-lang3</artifactId>
    </dependency>
[... 247 lines cut ...]
                </configuration>
              </execution>
            </executions>
          </plugin>
        </plugins>
      </build>
    </profile>
    <profile>
      
      <id>m2e</id>
      <activation>
        <property>
          <name>m2e.version</name>
        </property>
      </activation>
      <dependencies>
        <dependency>
          <groupId>org.apache.hadoop</groupId>
          <artifactId>hadoop-client-runtime</artifactId>
          <scope>compile</scope>
        </dependency>
      </dependencies>
    </profile>
  </profiles>
</project>