goecharger/go-eCharger-API-v2 #16: Access States

chk2902, 2021-09-20:
API1: "ast" war der "Access state", im Falle "ast=3" gab es Zeitbereiche zum Laden (undokumentiert), und so wie ich es verstanden hatte nicht gültig bei RFID oder Awattar (nicht sicher). Bitte korrigiere mich, wenn das nicht stimmt.

API 2 hat diesen Wert nicht mehr, sondern nur noch die Zeitbereiche.

Jeder der Zeitbereiche hat "disabled", "innerhalb" und "außerhalb" laut Doku. Wann sind bei API2 diese BEreiche aktiv? Immer, oder nicht bei RFID/Awattar?

chk2902, 2021-09-20:
Ach so, für Awattar gibt es einen Switch "use awattar" , für RFID nichts. Könntet ihr die Doku erweitern um die Info, wie man RFID forcieren kann (früher ast=1)? Geht dann ast=1 (RFID) und "use awattar" gleichzeitig (früher exklusiv, ast=1 oder ast=2)?

0xFEEDC0DE64, 2021-11-26:
Zeitbereiche aus dem Scheduler werden immer eingehalten. Wenn der ForceState abweicht von neutral, hat der auch noch Vorrang.
Ist der logic-mode=default und der scheduler erlaubt das laden zwischen 08:00 bis 14:00, und es ist aber schon 15:00, muss auch noch der force-state auf On gestellt werden, damit dann trotzdem geladen werden kann.