goecharger/go-eCharger-API-v2 #17: Solares Überschussladen

mlnoga, 2021-09-20:
Ich würde gerne solares Überschussladen mit dem go-eChargerumsetzen. Dazu scheint bisher zu fehlen:

- Wechsel zwischen 1-phasigem und 3-phasigem Laden per API. Der Schlüssel "fsp" ist als schreibgeschützt dokumentiert. Er sollte beschreibbar sein
- Alternativ: Niedrigere Ladeströme als 6A für den 3-phasigen Betrieb. Hier scheint ein hartes Limit vorzuliegen? Auch in der App ist kein niedrigerer Strom wählbar
- Genauere Regelung des Ladestroms als 1A
- Authentifizierung via API. Scheint undokumentiert

Über entsprechende Unterstützung würde ich mich sehr freuen.

bacanol, 2021-09-20:
Hey @mlnoga,

schau mal hier vorbei: https://github.com/evcc-io/evcc

Viele Grüße

Nico

mlnoga, 2021-09-20:
Schaut gut aus, Nico. Mein Wechselrichter ist von RCT. Der Hersteller hat mir freundlicherweise die Dokumentation des TCP/IP-Protokolls geschickt. Daher kann ich die aktuellen Leistungsdaten bereits auslesen. Habe ich praktischerweise auch in go gemacht. Vielleicht würde das ja ganz gut in evcc passen.

mlnoga, 2021-09-20:
Also, evcc funktioniert grundsätzlich, kann jedoch nicht automatisch zwischem einphasigem und dreiphasigem Laden umschalten. D.h., man muss sich manuell für den einphasigen Betrieb mit 1.4-3.7 kW für bewölkte Tage, oder für den dreiphasigen Betrieb mit 4.2-11 kW für sonnige Tage entscheiden. 

Das ist ja nicht so praktisch. 

Da die App die Einstellung der Phasenzahl erlaubt, muss es ja auch über ein API gehen. Nur welches?

mlnoga, 2021-09-20:
Also, `api/set?fsp=true` und `api/set?fsp=false` scheinen grundsätzlich zu funktionieren. Sie liefern zwar die Fehlermeldungen `{"fsp":"fsp: tried to set api key without setter (true)"}`  und `{"fsp":"fsp: tried to set api key without setter (false)"}` zurück. Jedoch verändert sich der in der App angezeigte Status von 1-phasig auf 3-phasig. Empfehlung: bitte hier dokumentieren und in der nächsten Firmwareversion die Fehlermeldung durch eine normale Statusrückmeldung ersetzen.

andig, 2021-09-21:
> Also, evcc funktioniert grundsätzlich, kann jedoch nicht automatisch zwischem einphasigem und dreiphasigem Laden umschalten. D.h., man muss sich manuell für den einphasigen Betrieb mit 1.4-3.7 kW für bewölkte Tage, oder für den dreiphasigen Betrieb mit 4.2-11 kW für sonnige Tage entscheiden.

Also evcc kann das schon. Nur hat Go-e es bisher nicht dokumentiert ;)

Sillium007, 2021-09-21:
@mlnoga : jemand anderes war fleißig am Testen: https://github.com/evcc-io/evcc/discussions/1547#discussioncomment-1360940

andig, 2021-09-23:
Siehe https://github.com/evcc-io/evcc/pull/1609 für vollständige evcc Unterstützung

andig, 2021-09-27:
Die Implementierung für evcc ist mittlerweile fertig und gemerged. Einschließlich Phasenumschaltung bei Leistungsüberschuss/mangel bei PV-geführter Ladung. mA-genaue Regelung kann die v3 allerdings auch weiterhin nicht...

mlnoga, 2021-09-27:
Top, Andi.

Ich hätte noch einen nativen golang connector für RCT PV, Batterie und Grid anzubieten. Interesse?

M

andig, 2021-09-27:
@mlnoga meld Dich bitte im evcc repo- das wäre cool. Im Moment machen wir das nur über das script plugin: https://github.com/evcc-io/config#meter-rct-power-storage-battery

closed 2021-09-27 by mlnoga