goecharger/go-eCharger-API-v2 #20: LEDs über API einstellen ?!

martipy, 2021-09-21:
Ich bekomme die LEDs über die API nicht gesetzt. 

`../api/set?cid="#00FF00"`  -->   `{"cid":"cid: tried to set api key with invalid json IncompleteInput (\")"}`

Habe schon alles mögliche probiert, aber leider ohne Erfolg. 

`../api/set?fna="test"`  (auch ein String Datentyp) wird problemlos übernommen.

Was mache ich falsch ? Was ist bei den LEDs anders ?

Firmware 051.4

Vielen Dank!

0xFEEDC0DE64, 2021-11-26:
Ab dem ersten # Zeichen wird von den meisten http client implementierungen der Rest der url einfach abgeschnitten, sieht man sogar schön im verbosen curl output:

```
$ curl -v 'http://192.168.0.47/api/set?cid="#00FF00"'
*   Trying 192.168.0.47:80...
* Connected to 192.168.0.47 (192.168.0.47) port 80 (#0)
> GET /api/set?cid=" HTTP/1.1
> Host: 192.168.0.47
> User-Agent: curl/7.80.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 500 Internal Server Error
< Content-Type: text/plain
< Content-Length: 74
< Connection: close
< Access-Control-Allow-Origin: *
< 
* Closing connection 0
{"cid":"cid: tried to set api key with invalid json IncompleteInput (\")"}
```

Lösung: die # als urlencoded schicken:

```
$ curl -v 'http://192.168.0.47/api/set?cid="%2300FF00"'
*   Trying 192.168.0.47:80...
* Connected to 192.168.0.47 (192.168.0.47) port 80 (#0)
> GET /api/set?cid="%2300FF00" HTTP/1.1
> Host: 192.168.0.47
> User-Agent: curl/7.80.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Type: text/plain
< Content-Length: 12
< Connection: close
< Access-Control-Allow-Origin: *
< 
* Closing connection 0
{"cid":true}
```

closed 2021-11-26 by 0xFEEDC0DE64

martipy, 2021-11-26:
Vielen Dank, funktioniert! Jetzt kann ich die Betriebsart (Netz oder Überschussladen) über die LEDs anzeigen lassen :-)