goecharger/go-eCharger-API-v2 #22: Modbus API Phasenumschaltung?

kaeferfreund, 2021-09-23:
Soweit ich die API Dokumentation angeschaut habe gibt es in der Modbus Dokumentation keinen Befehl für die Umschaltung zwischen 1p/3p.

Interessant ist die Modbus API, weil das Protokoll selbst deutlich weniger Overhead hat als die HTTP API. Bei unseren letzten Tests wurden die Modbus Anfragen von der V3 Box deutlich schneller beantwortet, als die HTTP Anfragen. Kann jemand diese Beobachtung bestätigen?

Wird die Modbus API in Zukunft weiter von go-e gepflegt und weiter entwickelt?

peterpoetzi, 2021-09-23:
Ja, wir arbeiten aktiv an der Weiterentwicklung der Modbus API

kaeferfreund, 2021-09-23:
klasse! Könnt Ihr noch was zu den zwei anderen Punkten sagen?
1. Performance verglichen mit der HTTP-API
2. Phasenumschaltung via Modbus in Zukunft geplant?

mincequi, 2023-07-01:
Hallo,
gibt es Neuigkeiten zur Phasenumschaltung per Modbus? @peterpoetzi 
Das wäre wirklich großartig.