goecharger/go-eCharger-API-v2 #23: amx setting not available in V2

albertov, 2021-09-23:
I have a hardware v3 go-eCharger with firmware v051.4 and noticed that you can no longer set the `amx` parameter with API v2 to set the current _without_ saving it into EEPROM. Only the `amp` parameter can be set. Besides this, if using API v1, when `amx` is set the value also affects `amp` and the change persists across reboots.

 I want to dynamically set the current as a function of the current output of a photovoltaic system but I'm worried that changing the current dynamically will prematurely wear out the EEPROM on the device. Will this be the case if `amp` is changed too often? Maybe hardware v3 doesn't suffer from EEPROM wear out?

peterpoetzi, 2021-09-23:
The V3 hardware does not have the concept of writing data at the same position on an EEPROM anymore. The number of writes is virtually unlimited, so it is not necessary anymore to have the amx parameter

closed 2021-09-23 by peterpoetzi

andig, 2021-09-23:
> The V3 hardware does not have the concept of writing data at the same position on an EEPROM anymore. The number of writes is virtually unlimited, so it is not necessary anymore to have the amx parameter

It could have been so simple to support this in a non-breaking way...

peterpoetzi, 2021-09-23:
Actually it does that in a non-breaking way. When you use the v1 api, that is still supported by the newest hardware V3, it still accepts amx. Only the new v2 api never supported amx so it cannot break anything.