goecharger/go-eCharger-API-v2 #27: 1p/3p-Umschaltung vor Ladestart

chk2902, 2021-09-30:
Da die meisten Boxen ja nur bis 16 A gehen, macht eine 1p/3p-Umschaltung ja keinen Sinn bei der PV_Ladung, da entsteht ja sonst eine Versorgungslücke (zwischen 16A 1p und 3*6A 3p).

Bleibt erst mal das Umschalten vor dem Start der Ladung, wenn man also z.B. ein 3-phasiges Auto laden möchte.

Kann man den entsprechenden Parameter gleich beim Einschalten ("acs=1&lmo=3&frc=2" (siehe https://github.com/goecharger/go-eCharger-API-v2/issues/10)) mitgeben? Oder muss das getrennt sein? Wie sehen die Kommandos aus (es gibt mehrere Register, die "phases" in der Beschreibung haben, alle bis auf "threePhaseSwitchLevel" sind R/O, und das ist ein float...)? Muss man irgendwas beachten mit Timing?

chk2902, 2021-09-30:
... und für später: wenn man das *während* einer PV-Ladung durchführen möchte, wie muss die Umschaltung 1p->3p und 3p->1p gemacht werden?

peterpoetzi, 2021-09-30:
Wie hier dokumentiert: https://github.com/goecharger/go-eCharger-API-v2/blob/main/http-de.md#werte-setzen kann man auch mehrere Werte setzen, indem man sie mit & aneinanderhängt (als GET Parameter)
z.B. http://192.168.0.75/api/set?acs=1&lmo=3&frc=2 eben

Das timing ist grundsätzlich egal, auch die App setzt alles auf einmal. Man kann alle Parameter auch während der Ladung verstellen

closed 2021-09-30 by peterpoetzi

chk2902, 2021-10-01:
@peterpoetzi bitte nicht schließen, das ist vorschnell...

"Wie sehen die Kommandos aus (es gibt mehrere Register, die "phases" in der Beschreibung haben, alle bis auf "threePhaseSwitchLevel" sind R/O, und das ist ein float...)? Muss man irgendwas beachten mit Timing?" ist unbeantwortet, und ich würde noch hinzufügen "was muss man vermeiden, damit die Auto-Ladegeräte keinen Schaden nehmen können"?

chk2902, 2021-10-01:
PS: "fsp" ist in der Doku "r/o". Der kann es also nicht sein.