goecharger/go-eCharger-API-v2 #30: Phasenumschaltung 1P/3P in API und Doku inkorrekt

chk2902, 2021-10-06:
"fsp" ist in der Doku r/o, und die API dafür (051.4) scheint konsequenterweise nicht zu funktionieren:

> http://192.168.2.59/api/set?fsp=false
Unexpected response code 500 for http://192.168.2.59/api/set?fsp=false
err: HTTP ERROR Internal Server ErrorServer-Fehler: {"fsp":"fsp: tried to set api key without setter (false)"} 

Interessantweise funktioniert es dennoch - man muss den Fehler nur ignorieren.

Interessant, dass das selbst bei "without setter" funktioniert. Wie geht das denn? Ich komme von der softwaretechnischen Seite, und ohne Setter geht nix (wenn man schon mal das Setter/Getter implementiert hat).

franky436, 2022-02-05:
Hab's bei mir gerade ausprobiert, folgendes funktioniert bei mir:

1-phasig:
http://192.168.1.102/api/set?psm=1

3-phasig:
http://192.168.1.102/api/set?psm=2

Wenn in der App "Ausstecken simulieren" aktiv ist, passiert das automatisch.

Joerg-Dr, 2022-02-23:
Kann es bestätigen: funktioniert!