goecharger/go-eCharger-API-v2 #37: eto funktioniert sei Update 051.8

martipy, 2021-10-31:
Mit dem Update auf **051.8** funktioniert bei mir der **eto** und **etop** Zähler nicht mehr richtig.

Der **wh** Zähler stimmt, aber **eto** zeigt viel zu kleine Werte (+300Wh statt +6,7kWh).

Mit den Downgrade auf **051.4** funktioniert eto wieder. Hardware ist die V3.

peterpoetzi, 2021-10-31:
Hi,

wie ist denn hier die Seriennummer? Gerne auch diskret an peter (at) go-e.co senden

Liebe Grüße,
Peter

copacetic82, 2021-11-01:
Hallo :) Vielleicht hängt mein Problem damit zusammen. Bei mir scheinen die Chip-zähler nicht richtig zu funktonieren, auch 051.8 und HW V3
- Ich habe für User 4 (unten mit ZZZZZ maskiert) einen chip angelernt, user 1-3 anfangs noch nicht.
- User 4 authentifiziert sich und lädt.
- Gesamtenergie zählt hoch (aktuell 10,58kWh geladen)
- Energie für User 4 zählt nur minimal hoch (aktuell 23 bzw. 0,02 kWh). 
- Bisher gab es generell noch nie unauthorisierte Ladevorgänge (nur von User 4 authorisiert), also müsste die User 4 Energie auch 10,58kWh entsprechen.
- Auto (eSmart) hat nur mit einer phase und max 1,8KW geladen, falls relevant.

Was ich bereits versucht habe:
- mehrmaliger Neustart per Sicherung bzw. ist der charger auch ein paar mal abgestürzt, sobald ich mir den verbrauch pro nutzer in der app anzeigen hab lassen. 
- mehrmaliges authorisieren bzw. mehrere ladevorgänge
- Chipzuweisung User 2 (XXXXX) und 3 (YYYYY)

{
    "name": "User 1",
    "energy": 0,
    "cardId": true
  },
  {
    "name": "XXXXX",
    "energy": 0,
    "cardId": true
  },
  {
    "name": "YYYYY",
    "energy": 0,
    "cardId": true
  },
  {
    "name": "ZZZZZ",
    "energy": 23,
    "cardId": true
  },
  {
    "name": "User 5",
    "energy": 0,
    "cardId": false
  },
  {
    "name": "User 6",
    "energy": 0,
    "cardId": false
  },
  {
    "name": "User 7",
    "energy": 0,
    "cardId": false
  },
  {
    "name": "User 8",
    "energy": 0,
    "cardId": false
  },

ddrrhh, 2021-11-04:
Ich kann das Phänomen bestätigen mit der Version 51.8 der Firmware funktionieren sowohl die energy Einträge bei cards als auch die Zähler eto und etop nicht mehr richtig. Der Zähler wh dagegen funktioniert. Meine Hardware ist V3

martipy, 2021-11-04:
Hatte Kontakt mit dem Hersteller. Der Bug wird mit dem nächsten Update behoben. Ich bin übergangsweise wieder auf die  051.4 zurück.

copacetic82, 2021-11-05:
Inzwischen wurde die 051.9 veröffentlicht. Upgrade hat bei meinen 3 chargern (HW V3) aber nicht geklappt. Stattdessen zeigt mir die App wieder 051.4 an, und erneute upgrades brechen ab...also Vorsicht. Ich habe es go-e gemeldet.

0xFEEDC0DE64, 2021-11-26:
Wurde mit 051.9 gefixt, das Update sollte auch für alle sichtbar sein

closed 2021-11-26 by 0xFEEDC0DE64