goecharger/go-eCharger-API-v2 #38: V3-HTTP1-API inkompatibel zu V2-HTTP-API mit "ast":"1": "alw" funktioniert nicht.

chk2902, 2021-11-02:
"ast=1"
"alw=1"

-> Box lädt nicht, "alw" bleibt 0.

Bei Hardware V2 funktioniert das so. Man kann dort "ast=1" lassen und mit "alw" ein- oder ausschalten.

Starter4711, 2021-11-03:
Ist bei mir gleich.
ast muss auf 0 gesetzt werden damit geladen werden kann, dann setzt aber die go oftmals einen Wert auf amx.

indivliving, 2021-11-19:
Bei mir das selbe. V3 Hardware kann mit keiner API Version oder MQTT den "alw" setzen. Gut laut API Key Dokumentation ist das ja auch nicht vorgesehen, in den Beispielen zu MQTT aber explizit aufgeführt das ganze über "alw" zu machen? Also irgendwo ist hier der Wurm drin.

chk2902, 2021-11-19:
Jetzt weiss ich endlich, was 'ne Wurmkiste ist... ;)

(https://wurmkiste.at/)

Spaß beiseite... leider wird einem nicht mitgeteilt, ob das mal korrigiert wird. Also nochmal langsam (ich habe mein eigens Problem erst mal nicht mehr verstanden...): auf der V3-Hardware funktioniert die HTT1-API nicht mehr so, wie sie auf V2 lief, ist also inkompatibel.

ECOM-Engineering, 2022-01-07:
### HW V3 und API V2: kann Laden nicht starten: 
**Mit Cloud API:**"alw" kann zwar gesetzt werden --> Rückmeldung
{"succeeded":[{"alw":true}],"failed":[]}
Aber status zeigt alw=false und charger lädt nicht

Mit WLAN (http) API wird code 500 zurückgegeben und charger lädt nicht.

Andere Werte, z.B. "amp" funktionieren.
Was mache ich falsch?