goecharger/go-eCharger-API-v2 #44: MQTT Topic entspricht ggf. nicht üblichen Standards

ako-3004, 2021-11-21:
im MQTT explorer ist mir aufgefallen, dass

0xFEEDC0DE64, 2021-11-26:
Ich fürchte, dass wenn wir den `/` am Anfang nun entfernen, werden ganz viele vorhandene Integrationen nicht mehr funktionieren und das wird zu Problemen führen. `/` am Anfang ist ja per Standard explizit erlaubt.

ako-3004, 2021-11-26:
Ich denke wenn ihr Euch das als Task hinterlegt das Topic langfristig für den User einstellbar zu machen, dann wäre das nicht schlecht. Aktuell sehe ich auch keinen Handlungsbedarf, wollte Euch den Sachverhalt aber mal melden.

nean-and-i, 2021-12-06:
Hallo,

bitte um Korrektur, es scheint ein zusätzlich leeres topic reingerutscht worden sein vmtl. durch doppel  "//"
Das dürfte ein Fehler sein, sowas macht keine einzige mqtt implementation.

[image]

Vielen Dank.

nean-and-i, 2021-12-06:
> Ich fürchte, dass wenn wir den `/` am Anfang nun entfernen, werden ganz viele vorhandene Integrationen nicht mehr funktionieren und das wird zu Problemen führen. `/` am Anfang ist ja per Standard explizit erlaubt.

leading / macht man nicht bei MQTT, man generiert dadurch nur mehr Probleme, wenn man zero value root topics erstellt.

Das ist schwerstens verpönt in der MQTT community -> Don’t use a leading slash (e.g. /mytopics ) to start your topic tree

thnx

peterpoetzi, 2021-12-09:
Wir werden mit Firmware 052.2 die Möglichkeit anbieten den Prefix für die mqtt topics setzen zu können.

Standardmäßig ist der Prefix exemplarisch mit der Seriennummer 123456:
"`/go-eCharger/123456/`". 

Das kann man anpassen auf "`go-eCharger/123456/`" wenn man nur den führenden / entfernen möchte, auf "`meineHomeAutomation/garageLinks/`", oder einfach auf einen leeren String "` `".

Der API parameter für mqtt topic prefix heißt `mtp`. Die Firmware 052.2 wird spätestens nächste Woche released, die App mit dem dazugehörigen Eingabefeld, kann eventuell etwas länger dauert, aber man kann den prefix ja auch über die API direkt setzen.

nean-and-i, 2021-12-10:
Vielen Dank.

Wird man über die go-e APP (iOS/Android) darauf einen Einfluss nehmen können?

Es wäre aber sinnvoller wenn per default der Präfix entfernt wird , damit es danach dem Standard folgt -> `go-eCharger/123456/` 

Das sollte eigentlich keine nennenswerten Probleme mit sich bringen, einmal eine bestehende integration gemäß standard umkonfigurieren (`/` entfernen), es gehen ja keine Daten dadurch verloren. (z.b. breaking changes passieren ständig bei homeassistant)

erxbout, 2024-12-17:
Dieses issue ist schon ziemlich alt, das topic ist mittlerweile in der App änderbar daher würde ich dieses issue schließen!

closed 2024-12-17 by erxbout