goecharger/go-eCharger-API-v2 #47: MQTT alw/set scheint nicht zu funktionieren obwohl MQTT Schreibzugriff erlaubt.

ako-3004, 2021-12-04:
Firmware 052
Aktiviere MQTT API "ON"
MQTT ohne Schreibzugriff "OFF"

Beim Testen der MQTT Anbindung fällt mir auf dass ich beispielsweise .../amp/set ausführen kann.
Jedoch .../alw/set geht nicht, hab auch /alw (ohne) set probiert.

[image]

[image]

[image]

[image]

[image]

Zum Schluss die Gegenprobe über die API v1 mit mqtt?payload=alw=1
Lädt wieder:

[image]

DaveMe81, 2021-12-12:
Hallo!
das ist mir auch aufgefallen. Ich denke, das alw im MQTT noch nicht setzbar umgesetzt wurde. In der Doku ist es aktuell auch nur als "Status" beschrieben.
[image]

peterpoetzi, 2021-12-12:
alw ist nur das Ergebnis einer komplizierten Berechnung und zeigt an ob das Auto jetzt geraden laden dürfte. Das hängt davon ab wie teuer der Strom gerade ist, wie spät es ist, ob die Ladestation überhitzt ist, etc.

Um einen Ladewunsch zu setzen sollte man frc (force state) nehmen und auf force-on oder force-off stellen.

ako-3004, 2021-12-12:
Ah okay, prima dann versuch ich einfach den frc...
Danke!

ECOM-Engineering, 2022-01-07:
> alw ist nur das Ergebnis einer komplizierten Berechnung und zeigt an ob das Auto jetzt geraden laden dürfte. Das hängt davon ab wie teuer der Strom gerade ist, wie spät es ist, ob die Ladestation [...]

frc=0: laden, frc=1: stop, aber was bedeutet frc=2? 
Die Beschreibung/Bedeutung des keys frc (forceState (Neutral=0, Off=1, On=2) versehe ich nicht

one4many, 2022-01-21:
Ich nehme mal an das frc=0 die kontrolle wieder an die interne go-e Logik zurück gibt (z.b. günstig laden bei eingeschalteter AWATTAR funktion)