goecharger/go-eCharger-API-v2 #50: MQTT - kein syntax check in der APP /  mqtt passwort ungeschützt

nean-and-i, 2021-12-10:
Nachdem es für die APP(s) kein github issue gibt, darf ich folgende Anliegen im Zusammenhang mit MQTT hier melden?

# mqtt - Bug request APP (iOS)

Zur Zeit ist es so, sobald man in der APP, MQTT aktiviert und im MQTT Feld einen Syntaxfehler erzeugt oder das Feld leer lässt, läuft der go-e in einem reboot loop und man lernt wie praktisch die Resetkarte sein kann.

# mqtt - Feature request APP(iOS): 

Eigene Eingabefelder für MQTT Benutzer und maskiertes Eingabefeld für Password, damit die MQTT Zugangsberechtigungen geschützt bleiben.

peterpoetzi, 2021-12-10:
Ich kann den Fehler mit dem Bootloop bei leerer bzw. syntaktisch falscher URL bestätigen. Wir werden in wenigen Minuten eine Firmware Update veröffentlichen das den Fehler behebt.

Bezüglich den Eingabefehlern für Benutzer und Passwort nehmen wir den Wunsch gerne auf, derzeit gibt es aber keine Timeline dafür.

peterpoetzi, 2021-12-11:
Die neue Firmware wurde released

closed 2021-12-11 by peterpoetzi