goecharger/go-eCharger-API-v2 #55: Was heisst strings „sollten“ umschlossen werden?

andig, 2022-01-02:
Was passiert wenn nicht? Bei url encoding ist das normalerweise nicht notwendig.

0xFEEDC0DE64, 2022-01-02:
der wert hinter dem = sollte JSON parsebar sein.

=abc ist kein json
="abc" hingegen schon
=123 auch parsebar
=true auch parsebar
=null auch parsebar
=1,3 nicht parsebar
=1.3 schon parsebar

andig, 2022-01-02:
Das ist ziemlich unüblich für ein Rest Api. Die Doku sollte da klarer sein- müssen statt sollen?

peterpoetzi, 2022-01-03:
Es gibt API keys die z.B. null oder einen String akzeptieren, z.B. mqtt broker URL

andig, 2022-01-03:
Leere Strings über Url (die sich als null interpretieren lassen) sind normalerweise genau das:

   foo=&bar=

d.h. also die Quotes für Strings sind zwingend weil sonst nicht passend zum Json?