goecharger/go-eCharger-API-v2 #58: API key "psm" Phasenumschaltung (fehlt in Dok)

ECOM-Engineering, 2022-01-21:
Die Phasenumschaltung per API erscheint nicht in der Dok. Einzig 'fsp' (read only) ist angegeben.
Durch Vergleich der Stati bei 1/3 phasig erscheint der geänderte Parameter 
**'psm' = 1 -> 1 Phase, 'psm = 2 -> 3 Phasen** (Umschaltung per App erzeugt).
Schreiben ist möglich per API und funktioniert auch. 

Ist die Umschaltung mit 'psm' per API der richtige Weg? 
Was bewirken die Zeiten 'psmd', 'mptwt' genau und werden diese bei Änderung mit 'psm' berücksichtigt?

franky436, 2022-02-05:
In der App gibt es unter Einstellungen->Auto->Ausstecken Simulieren mit zwei Zeiten: "Ausstecken simulieren Dauer" in Sekunden (erlaubte Werte: 5 - 3600 sec) und "Phasenumschaltung Dauer" (10 - 3600 s).

ECOM-Engineering, 2022-02-05:
Danke!

chk2902, 2022-02-09:
Achtung: "psm" erst ab Firmware "051.8", sonst "fsp" - ist r/w, auch wenn r/w in der Doku steht.