goecharger/go-eCharger-API-v2 #59: Feature req - set cardid / user

povlhp, 2022-01-22:
Would be nice to be able to set card ID thru API even when charger is in open mode. 
Say I use it at home and another location - would track power each location. If I could set cardID I could track power used each place without having to use a card every time. 
Would allow me to use charger to count kWh in 10 groups. 
Maybe use one for night charge. One for day charge.

Breale1, 2022-02-23:
Agree, would give a good possibility to seperate Grid and PV loaded energy as well! 
By the way, it seems as Cardmanagement via Mqtt is missing at all in version 052.02(eca to ec1, rca to rc1 and so on) !?

DkDavid, 2022-10-21:
Having multiple Go-E Wallboxen it would really be helpful to access (read/write) the RFID cards (Name and cardID). Currently I need to manually teach in a tag on multiple stations. Another option would be a API do to the authentication without using OCPP, which is quite overkill for most users but currently my way to go.

seh-len, 2024-12-25:
Gibt es hier schon einen Stand der Entwicklung?
Ich kann den Vorschlag nur unterstützen, gerade bei mehreren go-e Chargern ist es sehr umständlich, die gleiche Karte an verschiedenen Chargern anzulernen.
Ein erster Schritt wäre es, die jeweilige cardid im cards api key zur Verfügung zu stellen.