goecharger/go-eCharger-API-v2 #60: State remains in charging after forcing state to off

FarkiMon, 2022-01-28:
When you stop a charge from the http API by setting frc=1, the state of the car remains in "charging"("car":2) instead of "idle"("car":1) in the status. Is there any way to update this status without unplugging the car?

0xFEEDC0DE64, 2022-01-29:
What is the status of alw in that moment?

FarkiMon, 2022-01-29:
alw is set to false

FarkiMon, 2022-01-31:
I start the charge with /api/set?token=TOKEN&acs=0&lmo=3&frc=0 -> in the status I can see alw:true, car:2
When I want to stop the charge, /api/set?token=TOKEN&acs=0&lmo=3&frc=0 -> alw:false, car:2 (Expected car:1). The car status doesn't go back to car:1, and I can see in the app that the button says "Paused", until I unplug the car.

peterpoetzi, 2022-01-31:
It can take up to 6 seconds according to the Standard IEC 61851-1 for the car to stop charging after the charging is disallowed. Can you confirm that the car state also stays 2 after 6 seconds?

FarkiMon, 2022-01-31:
Yes, I have left it for long time over 15 min. I will run some more testing tomorrow.

closed 2022-02-03 by FarkiMon