goecharger/go-eCharger-API-v2 #61: API Key für "Geplanter Ladevorgang" fehlt

dako0815, 2022-02-04:
In der App kann man unter "Geplanter Ladevorgang" verschiedene Einstellungen tätigen. Für die Werte für "Lademenge" und "Fertig um" gibt es aktuell keinen API Key. Wird hierfür noch was kommen?
Ich würde nämlich gerne die Awattar-Logik vom go-e nutzen und nur die beiden Werte durch den iO Broker befüllen...

[image]

arboeh, 2022-02-04:
Für die Funktion muss der Geplante Ladevorgang auf der Startseite der App aktiviert sein. Key ist "lmo".
Mit dem Key "ate" kann man die "geplante" Energiemenge festlegen, mit "att" die Endzeit. Verwende ich in Home-Assistant und funktioniert bestens.

dako0815, 2022-02-04:
Ok perfekt. Vielen Dank. Das hatte ich leider nicht gefunden bzw. falsch interpretiert...

closed 2022-02-04 by dako0815