goecharger/go-eCharger-API-v2 #62: API return of "frc=0" inconstitent

mvo5, 2022-02-10:
First let me say THANK YOU for the nice API, the docs and the fact that I control my go-e via your APIs.

Firmware 0.53.1, local http APIv2

When setting the "frc" option to zero the json reply is inconsistent with the subsequent "status?filter=frc":
```
$ curl -v http://192.168.1.4/api/set?frc=0
{"frc":true}
$ curl http://192.168.1.4/api/status?filter=frc
{"frc":0}
```
It would expect a reply from "set" like: `{"frc":0}`.

It's also inconsistent with the (limited number I observed so far of) other API call returns where the json result matches the value that got set.

chk2902, 2022-02-10:
Nope. AFAIK the first reply is the success state (has been set successfully).

mvo5, 2022-02-10:
Uh, my bad then. Sorry

closed 2022-02-10 by mvo5