goecharger/go-eCharger-API-v2 #63: AMP/AMX sollte ein float sein

jw-rav, 2022-02-16:
Für eine PV-Regelung ist die Einstellung in ganzen Ampere-Schritten viel zu grob. Damit ist eine Ausregelung auf Null-Einspeisung nicht möglich.

Die ISO IEC 61851-1 erlaubt eine wesentlich feinere Einstellung als ganze Ampere-Schritte. Das sollte mindestens in 0.1A-Schritten oder noch feiner möglich sein, wie bei anderen Wallboxen auch.

franky436, 2022-03-28:
Interessanter Vorschlag. Habe aus Interesse mal Messungen in 5 Sekundenabständen gemacht. 

**Testszenario:**
go-e Charger (hw v3)
Model 3P 
Ladung: 16A / 1p (einphasig). 

Habe den "nrg" Wert über die lokale API v2 ausgelesen, in meinem Fall mittels:
**http://172.16.254.103/api/status?filter=nrg**

Hier die Watt-Werte die mir der go-e liefert. Pro Zeile ist jeweils 5 Sekunden später:

3810 W
3870 W
3850 W
3880 W
3830 W
3810 W
3890 W
3820 W
3800 W
3790 W
3800 W
3860 W
3870 W
3850 W

Fazit: ich habe 70 Watt Varianz innerhalb von nur einer Minute. Ich vermute es liegt schlicht daran, dass die Netzspannung leicht variiert. Bei 4 Volt mal 16 Ampere komme ich in die Gegend von 70 Watt. 

In dem Kontext stellt sich die Frage, ob man die Ampere wirklich so genau steuern können will/muss?

Möglicherweise ist es vom Ort/Stromanbieter/Tageszeit/Wochentag abhängig ...

chk2902, 2022-03-28:
Der Wunsch ist ja, dass man nicht in ganzzahligen Ampere-Schritten steuern muss, sondern wie z.B. der SmartWB in feineren (hier 1/100) Schritten. 

Dann bekommt man ein paar Prozent (8% bei 6A, 3% bei 16A) mehr Effizienz beim Nachregeln bei einer PV-Überschuss-Regelung. Eventuelle Abweichungen vom Stromstärke-Sollwert sind egal, da man ja eh ständig nachregelt. Weder ist der Eingangsspannung genau, noch die Auto-Ladeelektronik. Zudem ist diese auch nicht immer ganz linear, sondern hat Stufen - was für eine möglichst "feine" Abstufung bei der Wallbox spricht, damit die Stufen möglichst wenig "gegeneinander" laufen. Hier z.B. der Stromstärkeverlauf meines Nissan Leafs im Verhältnis zur "Soll"-Stromstärke im unteren Ampère-Bereich.

[image]

Resultat ist dann eine Ladekurve (in Blau), die schön zur PV-Leistung parallel läuft.

[image]

Im Gegensatz zum ganzzahligen Wert:

[image]

jw-rav, 2022-03-28:
> Vermutlich ist die Varianz vom Ort/Stromanbieter/Tageszeit/Wochentag abhängig. - Wie ist das bei Euch?

Weitere mögliche Quellen für die gemessene Schwankung:
- Messverfahren innerhalb des eCharger. Bei meinem Charger wird zumindest die Spannung nicht korrekt gemessen.
- Ladeelektronik des Autos
- Ungenauigkeiten bei der Generierung der PWM

Aber wie @chk2902 schon schrieb, ist die Schwankung bei Einstellung 16A nicht relevant. Ebenso irrelevant ist, ob der durch die PWM vorgegebene Strom-Wert exakt getroffen wird. Relevant ist lediglich, dass die PWM möglichst genau den Wunsch-Wert vorgibt und nicht schwankt. Nur dann hat das Auto auch die Chance, die Stromaufnahme stabil zu halten.

Man versucht ja, eine Null-Einspeisung zu erreichen. Deswegen verwendet man für diese Regelung ja auch nicht den im eCharger eingebauten (eventuell auch ungenauen) Zähler sondern einen am Hausanschluss installierten Zähler und regelt den eCharger anhand des gemessenen Wertes RELATIV nach. Wenn der eCharger zB momentan auf 13.02A steht und der Zähler sagt mir 0.5A Bezug, dann stelle ich den eCharger auf 12.52A. Ob das Auto nun aufgrund von irgendwelchen Ungenauigkeiten in diesem Moment tatsächlich 13.5A zieht oder 12.5A ist vollkommen gleichgültig.

Es kommt darauf an, fein granuliert RELATIV nachjustieren zu können. Und zwar so, dass der Zähler am Hausanschluss möglichst auf 0A geregelt wird. Damit werden auch automatisch sonstige Schwankungen im Hausnetz ausgeregelt (zB Einschalten eines Backofens).

Wenn man nur ganzzahlig nachregeln kann, dann hat man die Wahl zwischen Pest oder Cholera: Entweder bleibt man bis zu 230W im Bezug (und zahlt entsprechend viel für den Strom) oder man bleibt bis zu 230W in der Einspeisung (und muss den billig eingespeisten Strom später teuer wieder zurückkaufen). Bei dreiphasigem Anschluss sind es sogar bis zu 690W.

Ich regle im Sekundentakt nach. Schwankungen im Minutenbereich sind in diesem Szenario vollkommen uninteressant.

franky436, 2022-03-29:
Vielen Dank für Eure Einblicke! 

Ihr nutzt grundsätzlich eine andere Philosophie, die genauso Sinn macht, aber sich von meiner Strategie unterscheidet. In meinem Fall habe ich 3 Sekunden Zeitverzögerung vom Stromzähler. Bis die Wallbox dann nachregelt ist es oft schon zu spät.

Deshalb, und weil genug PV Strom vorhanden, ist meine Strategie die Spitzen zu reduzieren und den Headroom ganz bewusst zu nutzen. Einspeisen würde ich den verbleibenden Strom ja ohnehin wenn das Auto das Charge Limit erreicht hat.

Auch ein Faktor: Wir haben zB einen Induktionsherd der immer wieder ein/ausschaltet um verschiedene Stufen zu ermöglichen, ähnlich wie eine LED durch PWM die Helligkeit regelt. Er schwankt also sekündlich um bis zu 2000 Watt.

Mit der Ladelogik alle 1-Sekunden nachzuregeln würde ich aufgrund der Zeitverzögerung zwischen Stromzähler auslesen (bei mir satte 3 Sekunden Verzögerung!) und der Verzögerung die Wallbox nachregeln die Spitzen nur noch weiter verstärken, wie ein Oszillator.

Daher nutzt meine Ladelogik die Volatilität vom Hausverbrauch um Headroom für Spitzen zu schaffen. Hier ein Beispiel wenn meine Frau kocht und gleichzeitig das Auto lädt:

[image]

Ich gehe also ganz bewusst nicht bis auf 0 ran, sondern nutze den Headroom gezielt um die Spitzen abzufangen. Das macht natürlich nur Sinn, wenn die PV genug Strom produziert und das Auto entsprechend ein paar Minuten länger laden kann. - Wenn das Auto früher stoppt, würde ich den verbleibenden Strom ja trotzdem einspeisen.

jw-rav, 2022-03-30:
> Ihr strebt eine andere Ladestrategie an, die total Sinn macht und sich von meiner Strategie unterscheidet.

Ich denke nicht dass sich unsere Strategien wesentlich unterscheiden. Allenfalls kann ich nur unwesentliche Unterschiede bei der Umsetzung erkennen.

> Hintergrund: In meinem Fall habe ich 3 Sekunden Zeitverzögerung vom Stromzähler,

Was ist das für ein Zähler und wie ist der angebunden?

> und einen Induktionsherd der 2000W Volatilität verursacht.

Das ist ja nun etwas sehr spezielles. Unser Induktionsherd (BORA) macht sowas nicht. Auch von anderen Geräten habe ich sowas noch nicht gesehen.

> Deshalb, und weil (meist) genug PV Strom vorhanden, ist meine Strategie auch die Spitzen mittels PV-Ertrag abzudecken und den Headroom ganz bewusst zu nutzen.

Das kannst Du doch auch dann machen wenn die Einstellung feiner granuliert (also als Float) möglich ist.

> Einspeisen würde ich den verbleibenden Strom ohnehin wenn das Auto das Charge Limit erreicht hat (mal abgesehen von Dezember/Jänner).

Ja, genau der Winter ist das Problem. Je nach Grösse der Anlage auch die Übergangszeit. Immer dann wenn der Bedarf für das Auto grösser ist als die PV liefern kann musst Du jede voreilig eingespeisste kWh wieder teuer zurückkaufen. In dieser Zeit muss die Strategie lauten: Einspeisung auf jeden Fall verhindern.

> Der Induksionsherd schaltet immer wieder ein/aus um verschiedene Kochstufen zu ermöglichen, ähnlich wie eine LED durch PWM die Helligkeit regelt.

Wenn die LED mit einer PWM von 1/10Hz betrieben würde, dann würde ich diese dem Hersteller um die Ohren hauen (SCNR!-)

> Mit der Ladelogik alle 1-Sekunden nachzuregeln würde ich aufgrund der Zeitverzögerung zwischen Stromzähler auslesen (bei mir satte 3 Sekunden!) und der Verzögerung die Wallbox nachregeln die Spitzen nur noch weiter verstärken, wie ein Oszillator.

Ich sehe die Reaktion auf die Nachregelungen bereits beim nächsten Request. Das interpretiere ich so, dass innerhalb dieser Sekunde WB+Auto bereits reagiert haben und der Zähler die Änderung bereits erfasst hat. Wie ist denn Dein Zähler angebunden?

> Daher misst meine Ladelogik u.a. die Volatilität vom Hausverbrauch um Headroom für Spitzen zu schaffen.

Du passt diese Reserve also dynamisch an? Habe ich bislang noch nicht als erforderlich angesehen, weil ich hier solche Volatilität noch nicht beobachtet habe. Selbstverständlich kommt es immer wieder zu starken Schwankungen (bis zu 9kW bei der Sauna), aber diese sind innerhalb ein/zwei Sekunden wieder weggeregelt.

Ich bin mir jetzt auch nicht sicher wie ich Deine Grafik interpretieren soll. Es handelt sich ja hierbei um den Herd in Kombination mit der WB-Nachregelung. Ich kann hier nicht herauslesen wann der Herd schaltet, wann der Befehl zur Nachregelung an die WB geht und wann die Nachregelung in den Messwerten zu sehen ist. Aus dieser Grafik kann ich nicht herauslesen mit welcher Frequenz Dein Herd schaltet.

Oh, noch eins: schädlich sind ja nur die positiven Sprünge. Die negativen sind harmlos. (im  Sommer)

> Ich gehe also ganz bewusst nicht bis auf 0 ran, sondern nutze den Headroom gezielt um die Spitzen abzufangen.

Aber das ist doch kein Widerspruch?

Es geht hier doch nicht darum genau auf 0 zu kommen, sondern darum den gewünschten Zielwert (ob das jetzt 0W, +500W oder -500W sind) mit einer feineren Granularität einstellen zu können. Vorzeichen dieses Zielwertes ist abhängig von Jahreszeit und Betrag des Zielwertes ist abhängig von der Volatilität. So einfach kann es sein.

> Wenn kein Herd läuft und die Volatilität binnen ca. 1 Minute nachlässt, regelt das System hoch bis auf maximal 240W (also auf 1 Ampere genau).

240W über 10 Stunden sind eben auch 2.4kWh. Hier würde ich gerne auf einen einstelligen Wert kommen können. Positiver Wert im Sommer, negativer im Winter. Darum der Request, das in float einstellen zu können und dass die Box die PWM auch mit einer entsprechender Genauigkeit auch generiert.

Um Deien 3-sekündigen 2kW-Überhänge auf 240Wh aufzusummieren müsstest Du in dieser Stunde 144 Überhänge haben. Das ist ein Überhang alle 25 Sekunden. Angenommen, Dein Herd schaltet alle 12 Sekunden, dann kosten Dich die Überhänge bei einer Stunde kochen 480Wh. Demgegenüber stehen 2.4kWh bei einer Ladung von 10 Stunden ohne Kochen.

Selbstverständlich ist das eine stark vereinfachte Betrachtung. Diese soll aber nur einen Denkanstoss geben, ob die Prioritäten richtig herum sortiert sind. Als allererstes würde ich mich auf die Suche machen woher diese Verzögerungen kommen...

franky436, 2022-03-30:
Vielen Dank für den "Denkanstoß". 

Warum Du Float einstellen möchtest ist mir klar, danke. :)     

Ich habe das sehr genau durchdacht und im Laufe der 2 Jahre bereits mehrere Strategien implementiert. Der Unterschied ist: deine Strategie regelt nachträglich und ohne Logik für Headroom, daher ist für Dich die Zeitverzögerung so wichtig. Mit meiner Strategie ist eine kleine Verzögerung sekundär, denn meine Regelung arbeitet quasi "vorausschauend" auf Basis von kürzlich gemessenen Volatilitäten. Der so errechnete Headroom deckt mögliche zukünftige Spitzen ab. Nur die erste Spitze geht durch, danach regelt das System schon ein wenig runter und "macht Platz" für die eventuelle weitere Spitzen. Genau das ist der grundlegende Unterschied. Das mache ich ganz bewusst so weil ich meist ausreichend PV Strom zur Verfügung habe, daher ist mir wichtiger repetitive Spitzen möglichst vorab abzufangen, als die 0-Kennlinie zu erreichen und dafür zu riskieren dass immer wieder Spitzen im Netzbezug landen. - Das ist gewollte Strategie und nicht weil ich zu doof bin die 0-Kennlinie zu treffen. :)  Mein Umfeld mit zwei Haushalten ist vermutlich deutlich Volatiler als deines, daher verstehe ich dass Du es anders implementiert hast. Das heißt aber nicht, dass ich wenig nachgedacht hätte. :) 

In meinem Post wollte ich lediglich diesen Unterschied darlegen und warum ich es anders mache. Es ging nicht darum Dich davon zu überzeugen dass es in deinem Szenario besser wäre. Wenn bei Dir der PV Strom "knapp" ist und Du deshalb jedes Watt direkt ins Auto drücken willst, ist deine Strategie vollkommen legitim. Bei Verbrauchern mit hohen Volatilitäten halte ich das aber eben nur bedingt sinnvoll. Bei mir sind es 2000 Watt, in den oben geposteten Kurven vielleicht ein Zehntel davon.  Ist halt ein Unterschied.

Unser Induktionsherd ist auch ein BORA, genauer: BORA Basic.  Nur auf den beiden höchsten Kochstufen gibts keine Volatilität.    Die Induktionskochfelder von IKEA, die ich kenne, machen es auch so (PWM).    

Im Stromzähler ist ein Smart-Me Kamstrup Modul. Zeitverzögerung aufgrund deren Cloud, keine lokale API. Die Verzögerung ist wiegesagt weniger relevant wenn man mit Headroom arbeitet.

Da ich (meist) ausreichend PV Strom habe, ist die Frage nicht ob ich einspeise, sondern wann. - Daher haben wir auch unterschiedliche Ansätze. Ich habe Deinen Ansatz durchaus verstanden, und der ist vollkommen legitim. Heißt aber nicht, dass andere Ansätze schlechter sind oder ich zu wenig nachgedacht hätte. 

Schöne Grüße & frohes Überschussladen !!!

jw-rav, 2022-04-01:
> **Zu den Prioritäten:** Einspeisen hat einen Cost-Faktor von -6 Ct/kWh Netzbezug hat einen Cost-Faktor von 20 Ct/kWh

Das hängt doch stark ab von Höhe der Vergütung und der steuerlichen Betrachtung. Was für Dich Sinn macht muss für andere noch lange nicht ebenfalls sinnvoll sein. Wann genau welche dieser drei Strategien das beste Ergebnis liefert ist doch sehr individuell.

> Somit ist für mich klar, dass "Netznutzung minimieren" immer dann die Priorität sein sollte, sobald PV-Tagesertrag > Eigenverbrauch. Im Winter/bei Schlechtwetter ist "kein Einspeisen" besser, das habe ich abgedeckt durch ne Art Fuzzy-Logik, der bei "zu wenig" Tagesertrag auch Netzbezug zulässt.

Was ist mit dem Strom der eingespeist wurde BEVOR die Logik zuschlägt? Wie auch immer, das ist ein Nebenschauplatz...

Das Eine ist, eine Strategie zu haben (wie auch immer diese aussehen mag). Das ist NICHT das Thema dieses Tickets.

Das Andere ist, die gewählte Strategie möglichst genau umsetzen zu können. DAS ist das Thema dieses Tickets.

> Wenn man in dem Kontext alle drei Strategien betrachtet ist auch klar warum eine Ganzzahl/Integer für Ampere in vielen Fällen ausreicht.

Sehe ich anders. Aber wenn Du der Meinung bist dass für Dich 1A-Schritte besser sind, dann hindert Dich doch niemand daran eine Ganzzahl in ein Float zu schreiben?

Sorry, aber ich verstehe gerade nicht was genau Dein Problem ist und wieso Du so vehement versuchst, diesen Feature-Request wegzudiskutieren? Wenn Deine Regelung mit ganzen Werten besser funktioniert, kannst Du doch auch weiterhin Ganzzahlen verwenden?

> Wenn man zuwenig Strom hat, kann man 1 Ampere hochdrehen und ein paar Watt zukaufen um sämtlichen PV Strom bis aufs letzte Watt selbst nutzen zu können.

Nein, eben nicht! Ich will ja nicht mehr zukaufen als unbedingt nötig...

Ich habe hier eine Sauna, die PWM-Sprünge von 9kW macht. Willst Du vorschlagen genügend Headroom zu schaffen, dass diese 9kW-Sprünge abgefangen werden? Deine Fuzzy-Logik funktioniert doch sicher auch mit 10A-Schritten hervorragendst?

Ich halte es für die bessere Strategie den 9kW-Sprung zuzulassen und innerhalb von einer Sekunde wegzuregeln.

> **Zur 2,4 kWh Rechnung:** Bei Granularität von 1A / 240V = 240W speist man im Schnitt nur die Hälfte davon ein, also 120 W und nicht 240 W.

Ich wusste, dass das kommt. Genau deswegen hatte ich auch geschrieben, dass das eine stark vereinfachte Betrachtung ist! 

Die Kernaussage ist, dass Dein Headroom über die Zeit zu einem nicht mehr zu vernachlässigenden Betrag aufsummiert wird. Das muss man in Relation setzen zu den aufsummierten Überschwingern. Je schneller man in der Lage ist diese Überschwinger wegzuregeln, um so stärker fällt Dein Headroom ins Gewicht.

Man muss eben auch betrachten, dass die Ladezeiten wesentlich länger sind als die Kochzeiten. Das mag bei Dir ja ganz anders sein...

Je
1. grösser die Sprünge
2. häufiger diese Sprünge auftreten
3. langsamer diese Sprünge weggeregelt werden

um so mehr Headroom benötigst Du. Auf 1. und 2. hat man keinen Einfluss. Auf 3. hingegen sehr wohl!

Du hast die Prioritäten falsch herum sortiert: anstatt Headroom+Fuzzylogik solltest Du versuchen den Zeitverzug anzugehen. 

> Insofern erstaunt mich doch etwas, dass lt. Deiner Aussage kein Gerät in deinem Haushalt solche Spitzen aufweist. Anyways ...

Dass ich sowas noch nicht gesehen habe liegt vielleicht daran dass meine Regelung unmittelbar reagiert und nicht erst wenn die Daten drölf mal im LKW um den Globus gefahren wurden..

> Sobald fertig gekocht ist und die Volatilität nachlässt regelt das System fürs Laden nach und nach hoch bis es konstant an der 0-Kennlinie lädt.

???

Wie kommst Du an die 0-Linie wenn Du nur in 1A-Schritten einstellen kannst? Im Durchschnitt bist Du doch 120W NEBEN Deinem Zielwert...

> Der Unterschied bezüglich Spitzen und deren Auswirkung auf Netzbezug: Deine Strategie regelt nachträglich ohne Headroom auf die 0-Kennlinie. Daher ist für Dich die Zeitverzögerung so wichtig.

Umgekehrt wiird ein Schuh daraus:

Du MUSST Deinen Headroom schaffen, weil Deine Regelung einen erheblichen Zeitverzug hat.

Ich KANN ES MIR LEISTEN auf den Headroom weitestgehend zu verzichten, weil meine Regelung unmittelbar greift.

> Das dafür in Kauf genommene Einspeisen ist gewollt

Das magst Du ja gerne wollen. Erwarte aber doch bitte nicht, dass auch andere Deine sub-optimale Strategie übernehmen?

ICH will voreilige Einspeisung vermeiden! ICH will keinen Headroom, da das eine VOREILIGE EINSPEISUNG ist!

> Die erste Spitze geht durch, danach regelt das System schon ein wenig runter und "macht Platz" für die eventuelle weitere Spitzen. Nach 3-4 Spitzen hat er genau den 2000W Headroom, den ich für den Induktionsherd haben will, sodass die Spitzen unter der 0-Kennlinie bleiben (bzw. im Optimalfall möglichst genau daran grenzen).

Und ich kann es mir leisten nahezu komplett auf den Headroom zu verzichten, weil Sprünge unmittelbar weggeregelt werden. Warum sollte ich denn unnötig genau diesen Headroom voreilig einspeisen?!? Das macht doch keinerlei Sinn!

> Im Stromzähler ist ein Smart-Me Kamstrup Modul. Zeitverzögerung aufgrund deren Cloud, keine lokale API.

Naja, wenn man über Cloud/App/SonstigeModerneBloatWare geht, dann muss man sich über diese Verzögerungen auch nicht wundern.

Ich befrage den SMA-Homemanager direkt, und habe über den gesamten Regelkreis Reaktionszeiten unterhalb einer Sekunde.

Würde ich über die SMA-Cloud gehen, hätte ich Verzögerungen von 15 Minuten(!) bis zu 12 Stunden(!). Willst Du vorschlagen Deine Fuzzy-Logik darauf zu erweitern?

> Die Verzögerung ist wiegesagt weniger relevant wenn man mit Headroom arbeitet.

Wie schon oben erwähnt, würfelst Du Ursache und Wirkung dircheinander: Die Verzögerungen ZWINGEN DICH zu diesem Headroom. Hättest Du keine Verzögerung, könntest Du auf den Headroom und die dadurch verursachte voreilige Einspeisung verzichten.

Dieser Headroom ist GENAU DIE VOREILIGE EINSPEISUNG, die ich vermeiden will.

franky436, 2022-04-03:
Ich weiß, dass Du Einspeisung vermeiden willst. Ich habe Deine und meine Strategie klar definiert in meinem Post, und wann und warum welche Strategie in welcher Situatoin besser ist.

Ich bleibe dabei: Du optimierst weitestgehend den Corner Case. Warum habe ich oben erklärt.

Und nein, die Zeitverzögerung ist nicht der Grund. Nur weil Du deine Spitzen nicht siehst, weil dein Messintervall 1 Sekunde ist, sind die trotzdem noch da. :)

jw-rav, 2022-04-03:
> Dass Du das so möchtest, habe ich verstanden. Dass es in 2 von 3 Fällen optimierte Strategien gibt die kein Float brauchen, auch.

Das ist Deine Behauptung, die jeder Grundlage entbehrt...

Was voreilig eingespeist wurde kannst Du nicht mehr zurückholen. Da hilft Deine Fuzzy-Logik auch nicht weiter...

> Dass viele diese nicht implementieren (weil komplexer)

Komplexität ist nicht das Problem. Das Problem ist, dass man nicht mehr zurückholen kann was voreilig eingespeist wurde.Dein Headroom ist genau diese voreilige Einspeisung.

> Dass Spitzen durchgehen bei egal wieviel Sekunden Verzögerung, auch.

Ich habe nicht behuptet dass Spitzen nicht durchgehen.

Die Auswirkungen werden aber vernachlässigbar, wenn man diese schnell genug wegregelt.

franky436, 2022-04-03:
In meinem Post habe ich sachlich dargelegt, dass man unterscheiden muss ob die Tages-PV-Produktion größer, gleich, oder kleiner dem Eigenverbrauch ist.

Leider bist Du nicht gewillt, diese Differenzierung nachzuvollziehen. Du hast einen Fall im Kopf (Tages-PV-Produktion < Eigenverbrauch) und für den Fall stimmt deine Argumentation ja auch, aber eben nicht für die anderen Fälle. 

Wie auch immer. - Schönen Sonntag noch!!

franky436, 2022-04-03:
Neuer Versuch: 

Netzbezug kannst Du nicht mehr zurückholen. Kostet 20 Cent / kWh,  später eingespeist: 20-6 = 14 Cent / kWh.
Eingespeist kannst Du nicht mehr zurückholen. Kostet -6 Cent / kWh,  später nachgekauft:  20-6 = 14 Cent / kWh.

Du kaufst tagüber *sicher* die Spitzen um 20 Cent / kWh anstelle *vielleicht* abends um 14 Cent nachzukaufen zu müssen.

Der Clou:
Wir reden hier vom PV-Überschussladen, sprich mein Auto lädt abends nicht und kaufe dafür somit keinen Strom nach.  Stattdessen lade ich einfach am nächsten Tag weiter.

franky436, 2022-04-03:
Netzbezug: 20 Ct/kWh
Einspeisen: -6 Ct/kWh

Wenn Du noch sachlich darlegen könntest, wo und wie Du den Netzbezug für deine Spitzen zurückholst ?

franky436, 2022-04-03:
*Beides* kann man nicht zurückholen:

Einspeisen -6 Ct / kWh
Netzbezug: 20 Ct / kWh

Finde den Fehler.

franky436, 2022-04-03:
Die Grundlagen habe ich oben beschrieben. Aber schon klar, Du willst einfach nichts anderes hören als deine festgesetzte Meinung. - Habe ich zur Kenntnis genommen.

mincequi, 2023-10-22:
Gibt es hierzu ein Update?
Laut Spezifikation sollte eine Regelung in 0.6 Ampere Schritten möglich sein.