goecharger/go-eCharger-API-v2 #64: undocumented frc values 3, 4 and 5

mze9412, 2022-02-28:
My box gives me values for frc that are not documented. This is values 3, 4 and 5. Has anyone seen those before?

[image]

0xFEEDC0DE64, 2022-04-11:
This can happen by setting invalid values through the api, every unknown value will be treated as "neutral". In a future firmware version we will implement proper value checks to not allow setting 3, 4, 5, ... anymore

mze9412, 2022-04-11:
> This can happen by setting invalid values through the api, every unknown value will be treated as "neutral". In a future firmware version we will implement proper value checks to not allow setting 3, 4, 5, ... anymore

It happens when setting lmo. After setting lmo - i.e. to 3 - frc sets itself to an undocumented value.