goecharger/go-eCharger-API-v2 #66: 5A support bei 3p Ladung

franky436, 2022-03-07:
Nicht alle E-Autos unterstützen 5A Ladung, das ist mir bewusst. Die mit am meistverkauften Fahrzeuge (Tesla) jedoch schon.

Beim PV Überschussladen habe ich aktuell 1p 16A und 3p 6A ein relativ großes Gap. Angesichts der typischen PV Anlagen Dimensionierung bei Einfamilienhäusern (oft rund um die 5 kWp) ist das genau in dem Bereich, wo man aufgrund des 6A Minimums gezwungen ist relativ lange auf 1p zu laden obwohl man eigentlich schon mit 3p 5A laden könnte.  Die 3p 6A sind hingegen kaum zu erreichen, das erreiche ich nur unter Optimalbedingungen mitten im Sommer.  Also ladet man überproportinal häufig einphasig.  Zumindest bei mir ist das so, noch dazu habe ich eine relativ lange Zuleitung was im hohen Amperebereich zusätzlich Einbußen mit sich bringt.

Im Sinne der Schieflast, Netzentlastung und PV-Überschussladung wäre es wirklich klasse, wenn wir auch 5A beim Ladestrom setzen könnten, ggf. nur nach Freischaltung auf Kundenwunsch z.B. für Teslafahrer mit PV-Überschussladung.

Vielen Dank!

martiby, 2022-03-07:
Zu 100% exakt auch mein Wunsch !!!  Müsste auch gar nicht in die App, **amp=5** über die API für den Regler würde völlig ausreichen. Vielen Dank!

stumbaumr, 2023-03-29:
Äh, das ist immer noch im Status Open - bei mir mit evcc.io über die APIv2 geht das schon bis 3x 5A runter.

Allerdings würde ich das doch gerne über die API bis 3x 1A runterregeln können. In der Tesla API geht das.

Hintergrund: Wenn man nur über PV lädt und da kommen mehrere große langsame Wolken daher:
- Wechsel auf 1-phasig (es wird der Schütz im Auto aus-angeschalten)
- Aus  (es wird der Schütz im Auto aus-angeschalten)
- An  (es wird der Schütz im Auto aus-angeschalten)
- Wechsel auf 3-phasig (es wird der Schütz im Auto aus-angeschalten)

Wenn ich bei einer solchen Wetterprognose mit 3x 1A Laden könnte würde ich mit Min+PV Laden:
- Von 3x 16A auf 1A runterregeln und mit 690W aus dem Netz weiterladen
- Wieder auf 11kW hochregeln
- Und der Schütz ging nie an oder aus...

Klar - das geht mit Tesla über die API - keine Ahnung ob das mit dem Go-E und dem Tesla über den Ladestecker geht...

franky436, 2023-04-06:
Vielen Dank für den Hinweis!

Tatsächlich geht es runter bis 1A mit https://github.com/tdorssers/TeslaPy ... eben ausprobiert!

peterpoetzi, 2023-04-06:
Der Standard für Typ2 laden, IEC 61851-1, definiert leider nur einen PWM Wert für Stromwerte ab 6A.

Natürlich steht es dem Auto frei auch einen kleineren Ampere-Wert zu ziehen, nur kann die Ladestation dem Auto eben nicht mitteilen dass weniger als 6A zur Verfügung stehen. Daher muss man ein Limit von weniger als 6A zwingend am Fahrzeug einstellen.

Siehe dazu auch der [Wikipedia Artikel](https://de.wikipedia.org/wiki/IEC_62196_Typ_2):

- **In der Norm IEC 61851-1 definiert der Bereich von 8 % bis kleiner 10 % Duty cycle die maximal verfügbare Leistung als 6 A,** 
- der Bereich von 10 % bis 85 % Duty cycle die maximal verfügbare Leistung als (% Duty cycle) multipliziert mit 0,6 A [...]. 
- Zwischen 0 % bis 3 % Duty cycle, 7 % bis 8% Duty cycle und 97 % bis 100 % Duty Cycle ist Laden nicht erlaubt. 
- Zwischen 3 % bis 7 % Duty Cycle wird signalisiert, dass eine digitale Kommunikation [...] das Laden regelt.

franky436, 2023-04-15:
In Ordnung. Für Tesla Fahrer somit folgende Lösung:

1. https://github.com/tdorssers/TeslaPy installieren
2. python cli.py -e deinteslakonto@gmail.com -c CHARGING_AMPS -k charging_amps=4 2>&1

Wobei **charging_amps=4** für 4 Ampere steht, einstellbar von 1 bis 16 Ampere für Model 3 ...

erxbout, 2024-09-26:
Funktioniert deshalb, weil die Ladebox zwar einen Strom vorgibt, allerdings das Auto entscheidet wieviel es ziehen möchte ;)

closed 2024-09-26 by erxbout

franky436, 2025-02-25:
No shit, Sherlock ;-)

franky436, 2025-02-25:
> Funktioniert deshalb, weil die Ladebox zwar einen Strom vorgibt, allerdings das Auto entscheidet wieviel es ziehen möchte ;)

Das war denke ich jedem von Beginn an klar. Aber vielen Dank für deine Antwort ;-)