goecharger/go-eCharger-API-v2 #67: Home Assistant auto discovery

Rothammel, 2022-03-12:
ich würde Home Assistant MQTT V3 auto discovery topics begrüßen.

syssi, 2022-03-23:
@Rothammel Bis sich der Hersteller die Mühe gemacht hat jeden einzelnen Sensorwert sauber für das Auto-Discovery zu beschreiben, möchte ich auf meine Custom-Component aufmerksam machen: https://github.com/syssi/homeassistant-goecharger-mqtt

0xFEEDC0DE64, 2022-03-25:
Ich habe damals vor längerer Zeit versucht, die http-api vom go-e charger ordentlich in homeassistant zu integrieren, leider fehlte mir schlussendlich die zeit und beim pull-request haben die alle möglichen requirements aufgestellt, von denen ich anfangs nichts wusste.
Falls wer homeassistant von source "bauen" und ausführen möchte, hier ist mein fork noch zu haben:
https://github.com/0xFEEDC0DE64/homeassistant-core

ahochsteger, 2022-04-22:
@Rothammel, @syssi, @0xFEEDC0DE64, vielleicht ist das für euch von Interesse:
https://github.com/joscha82/wattpilot/pull/9

Rothammel, 2024-06-27:
wow, sehr gut.
mir persönlich würde:
- amp/set
- att/set
fehlen.
was ich noch verwende ist der /car Status, das ist aber eine number und muss dann sowieso umgemapt werden

erxbout, 2024-07-10:
Den carstate mappen wir:
[image]

amp set: (also change auf das entity in homeassistant müsste das normal triggern)
[image]

Ich weiß nicht ob att zu spezifisch ist zum noch dazuschmeißen..
Aber das wäre durch eine automatisierung relativ leicht machbar..
In homeassistant einen helper mit dem entsprechenden wert und wenn dieser sich ändert dann aufs mqtt damit:

``` yaml
alias: Set_Goecharger_ATT
description: ""
trigger:
  - platform: state
    entity_id:
      - input_number.<thecreatedhelper>
    for:
      hours: 0
      minutes: 0
      seconds: 10
condition: []
action:
  - service: mqtt.publish
    data:
      qos: "0"
      topic: goecharger/att/set
      payload: "{{ states('input_number.<thecreatedhelper>') | int }}"
  - service: mqtt.publish
    data:
      qos: "1"
      topic: BalkonyMKII/TARGETOVR
      payload: "{{ states('input_number.<thecreatedhelper>') | int }}"
mode: single

```

erxbout, 2024-09-26:
@Rothammel würde das issue dann schließen

closed 2024-09-30 by erxbout