goecharger/go-eCharger-API-v2 #69: Bekommt man auch irgendwie den Akkustand zurück

fohnbit, 2022-03-17:
Hallo!

Ist es möglich den aktuellen Akkustand des Autos abzurufen?

Das wäre mehr als Interessant zu wissen.

Danke!

peterpoetzi, 2022-03-17:
Leider ist das über den Typ2 Stecker derzeit nicht möglich.

fohnbit, 2022-03-17:
Ok danke ... Und könnte man, obwohl 3phasig angeschlossen, auch auf 1phasig per API umschalten?

Oder muss ich dafür auf eine Steckdose umstecken?

peterpoetzi, 2022-03-17:
Das Umschalten zwischen 1-phasig und 3-phasig ist auch während dem Ladevorgang möglich.

fohnbit, 2022-03-17:
Danke!

jw-rav, 2022-03-28:
Iso 15118 erweitert die Spezifikation des Typ2-Steckers insofern, als dass auf Phase1eine zusätzliche Kommunikation über Powerline erfolgen kann. Hier kann dann auch der SoC ausgelesen werden.

Aktuelle Fahrzeugmodelle unterstützen das bereits (zB Hyundai Ioniq und Ioniq 5).

peterpoetzi, 2022-03-28:
> > Das Umschalten zwischen 1-phasig und 3-phasig ist auch während dem Ladevorgang möglich.
> 
> Gerüchten zufolge mögen das allerdings nicht alle Fahzeug-Modelle. So sollen einige ZOE's darauf mit Rauchzeichen aus der Ladeelektronik reagiert haben.

Was die ZOE nicht mag, ist dass sie nur auf Phase 1 und 3 mit Spannung versorgt wird. Hier scheint die Logik im Onboard Charger so zu sein, dass wenn auf Phase 2 keine Spannung festgestellt wird, der Anschluss wohl einphasig sein könnte. Der Onboard Lader verbindet dann N mit L3 (ohne geprüft zu haben ob auf L3 Spannung ist) und zerstört sich selbst.

Das kann mit dem go-eCharger nicht passieren, da dieser die Spannungen auf allen Phasen prüft. Auch das umschalten von 1- auf 3-phasiger Ladung während dem Ladevorgang ist mit dem go-eCharger mit der ZOE unproblematisch möglich.

fohnbit, 2022-03-29:
> Iso 15118 erweitert die Spezifikation des Typ2-Steckers insofern, als dass auf Phase1 eine zusätzliche Kommunikation über Powerline erfolgen kann. Hier kann dann auch der SoC ausgelesen werden.
> 
> Aktuelle Fahrzeugmodelle unterstützen das bereits (zB Hyundai Ioniq und Ioniq 5).

Puh, das wärs ... man könnte dann das Laden optimieren.

jw-rav, 2022-03-30:
> Was die ZOE nicht mag, ist dass sie nur auf Phase 1 und 3 mit Spannung versorgt wird. Hier scheint die Logik im Onboard Charger so zu sein, dass wenn auf Phase 2 keine Spannung festgestellt wird, [...]

Da der eCharger Phasen 2 und 3 wegschaltet, stellt das wohl kein Problem dar. Da gibt es auch keine Racing-Condition während der Umschaltung dabei?

Das heisst, die Umschaltung kann ohne Wegnahme des CP erfolgen? Die Ladung wird also nicht unterbrochen und es erfolgt lediglich ein Sprung bei der (3phasig saldierten) Stromaufnahme?

closed 2024-06-28 by erxbout

Boldfor, 2024-08-17:
> Iso 15118 erweitert die Spezifikation des Typ2-Steckers insofern, als dass auf Phase1 eine zusätzliche Kommunikation über Powerline erfolgen kann. Hier kann dann auch der SoC ausgelesen werden.
> 
> 
> 
> Aktuelle Fahrzeugmodelle unterstützen das bereits (zB Hyundai Ioniq und Ioniq 5).
> 
> 

Kurze Frage hierzu: Eine öffentliche Säule hat mir heute den SOC angezeigt, d.h. mein Fahrzeug beherrscht das weitergeben des SOC. Soweit ich verstehe, unterstützt aber erst der go-e charger Pro die ISO 15118, die das ermöglicht, richtig?

0xFEEDC0DE64, 2024-08-19:
Der go-e charger Pro mit ISO 15118 ist hardware-technisch dafür bereit, später ein Software Update von uns zu bekommen, dass dann schlussendlich wirklich ISO 15118 mit dem Auto spricht und damit auch den SOC bekommt (wir könnten den zb einfach als api key rausführen).
Bis das Software Update für den Pro verfügbar ist, spricht dieser gleich wie jeder andere go-eCharger nur mittels CP Signal zum auto und da gibt's keinen SOC leider

Boldfor, 2024-08-19:
> Der go-e charger Pro mit ISO 15118 ist hardware-technisch dafür bereit, später ein Software Update von uns zu bekommen, dass dann schlussendlich wirklich ISO 15118 mit dem Auto spricht und damit auch den SOC bekommt (wir könnten den zb einfach als api key rausführen).

Jene ISO 15118-Funktion des Pro-Chargers könnte ein Problem durch die Hintertür lösen, welches (meines Wissens nach) jeder Nutzer eines eAutos aktuell hat, das in einer Tiefgarage steht: Häufig gibt es nämlich dort keine Mobilfunk-Abdeckung, die benötigt wird, um den SOC über die einschlägigen Apps abzurufen (zumindest bei VW braucht's hierfür nämlich 5G, WiFi im Keller bringt gar nix und führt nur zu Webradio im Auto, die Kernfunktionen benötigen vermutlich aus Sicherheitsgründen Mobilfunk).

Da sich WiFi deutlich leichter in den Keller bringen lässt (Powerline, Repeater, etc) wäre die Pro ein einfacher Alternativ-Weg, an den SOC zu kommen.

Über genau jene Route (go-e charger via WiFi) stelle ich aktuell auch eine Möglichkeit zum gesteuerten Laden her, da auch hierfür (das könnte nämlich natürlich auch die App des Auto trigger) WiFi beim Auto nicht reicht, beim go-e charger schon.

Also: Danke an go-e für die baldige Möglichkeit. Bleibt zu hoffen, dass der pro-Charger preislich im Rahmen bleibt.

schinagl, 2024-08-19:
> Da sich WiFi deutlich leichter in den Keller bringen lässt (Powerline, Repeater, etc) wäre die Pro ein einfacher Alternativ-Weg, an den SOC zu kommen.
> 

Und ein CATx Kabel runterziehen?
Die go-e pro soll eine Ethernet Schnittstelle haben.... 
Bin a bissi oldschool was power line und wifi betrifft..... das Teufelszeug hat soviel das schief gehen kann😇

Boldfor, 2024-08-20:
> Und ein CATx Kabel runterziehen?

Das wäre die beste Option, ja. In typischen MFH ist das aber oft nicht möglich (zu viele Stahlbeton-Wände und Decken dazwischen) und oft ja auch gar nicht erlaubt (soll ja nicht jeder an der Substanz rumbohren können). Und wenn wlan auch nicht geht (hier: schon eine Stahlbeton-Decke führte zu nahezu 100% Abschirmung) ist Powerline die einzige Möglichkeit, um ein Signal vom Stellplatz in die eigene Wohnung zu bekommen. Wobei AVM Powerline erstaunlich leistungsfähig: Recht stabile Verbindung, trotz nahezu 100 Meter Leitungslänge. Wichtig: Beide Steckdosen müssen auf der gleichen Phase sein, ein Phasenübersprung geht auf die Entfernung nicht mehr.

schinagl, 2026-09-28:
> Ist schon bekannt, über welche APIs go-e die ISO 15118 Funktionalität zur Verfügung stellen wird? HTTP? MQTT? Modbus TCP? OCPP? Alle?

Da muss ich nachhaken, da schon ein Jahr vergangen ist.
Hab nun eine go-e pro, und die App lässt mich mit Version 60.5 'optional' ISO 15118 einschalten. (mein Auto kanns

Gibts Api Key zum Auslesen des PlugAndChargeIdentifiers? ( und nicht nur via ocp)
Und Auslesen des SOC des Autos? Der wär echt interessant!

Und wenns die Keys gibt... verratet ihr sie uns? ;-)