goecharger/go-eCharger-API-v2 #70: "lbp" liefert unplausible Werte

naerschhersch, 2022-03-22:
Hallo,

entweder handelt es sich um einen Verständnisfehler meinerseits oder um eine defekte Funktion.
Eine Abfrage von lbp sollte in meinem Verständnis (und nach Doku) die Länge des letzten Knopfdruckes in ms ausgeben. Der letzte Wert, den meine Box nach einem Knopfdruck ausgegeben hat, welcher ca. 3s lang war, lautete "168444283". Er sollte in der Größenordnung 3000 liegen, wenn ich es richtig verstanden habe.

Verständnisfehler oder Bug?

naerschhersch, 2022-04-21:
Gibt es hierzu Neuigkeiten oder Hinweise, wie ich sonst tracken kann, ob/wie oft/lange der Knopf gedrückt wurde?

peterpoetzi, 2022-04-21:
Der `lbp` api key gibt an zu welcher Systemzeit (in Millisekunden seit dem Reboot) der Knopf gedrückt wurde. 
Den Wert kann man von rbt abziehen, dann bekommt man heraus vor wie vielen Millisekunden das war.

Einen api key für wie lange der Knopf gedrückt wurde gibt es noch nicht. Ich denke wir können das aber zeitnah umsetzen und als lbpd (last button pressed duration) verfügbar machen.