goecharger/go-eCharger-API-v2 #73: MQTT API URL - special characters in password/username

thelukasssheee, 2022-04-13:
Hey everyone,
just installed my V3 go-e fix charger and tried to get MQTT up and running.

Using format mqtt://USER:PASS@ADDRESS:PORT

It seems, the app is always return the MQTT URL to the default value (mosquitto test server) whenever I'm entering my actual password. 
Narrowing it further down, this seems to be linked to my password containing special characters, e.g. "#".

How can I solve this (preferably, without having to change my password for all devices)?

Is there a way to type in the # in a different way, e.g. ASCII codes?

Any hint is welcome. Thanks 👍

syssi, 2022-04-13:
You could add a second user to your MQTT broker using a password without special chars.

thelukasssheee, 2022-04-13:
Great idea! Haven't thought of that! Will try this out and come back to you as soon as I've tested it.

syssi, 2022-04-13:
If you don't use ACLs every user is able to read & write everything.

thelukasssheee, 2022-04-13:
Okay! Just checked it. I can confirm, special characters was the issue. With an altered password and a new user, go-eCharger finally appears in MQTT (strangely, though, as a sub-topic of an empty root topic?!). 

I would leave it up to administrators to close the ticket or leave it open (in case you can/want to allow special characters in future versions). I would recommend, though, to add this information into the MQTT description: format of the URL string and allowed/not-allowed characters. 

From my side, issue is fixed with this workaround.

syssi, 2022-04-13:
If you ask me: Please don't close the issue. It should be handled properly IMO.

frog60, 2023-02-19:
I tried to to start publishing with GO-eCharger to Home Assistant / mosquitto broker (2.0.11). The URL is built using mqtt://USER:PASS@ADDRESS:PORT (USER and PASS use only letters!), but I cannot see any published items in MQTT Explorer. Using mosquitto_pub -u USER -P PASS it works and I can see the published items. What went wrong?

GO-eCharger: Firmware 055.0