goecharger/go-eCharger-API-v2 #76: Überschussladen <-> Maximalstrom-Laden

eydam-prototyping, 2022-04-24:
Hallo zusammen,

Wenn ich das richtig verstanden habe, ist für das Überschussladen der MQTT-Key ama vorgesehen, richtig?
Ich trage hier also ein, wie viel Strom meine Solaranlage gerade pro Phase bereitstellen kann (mind. 6A, max. 16A).

Gibt es in der App eine Möglichkeit, zwischen Überschussladen und Maximalstrom-Laden umzuschalten? Wenn keine Sonne scheint, möchte ich mein Auto ja trotzdem irgendwie laden. Und dafür mein Skript anhalten, was diesen MQTT-Key beschreibt, ist etwas unpraktisch. Oder wie lösen andere dieses Problem?

Viele Grüße
Tobias

deistho, 2022-12-04:
Stehe vor dem gleichen Problem, verbinde mich jedoch über Modbus. Wenn man auslesen könnte, welcher der bis zu 5 Lademodi gewahlt ist, könnte man sich z.B. den ersten für reines Überschussladen hernehmen, den zweiten für min. + Überschuss, und den dritten für max Power. Kann der Modbuskanal noch hinzugefügt werden?

Knedox, 2022-12-29:
same problem. 
Da muss das steuer script wohl separat deaktiviert werden...