goecharger/go-eCharger-API-v2 #77: dll - link funktioniert nicht

Ventusfahrer, 2022-04-25:
wir haben zwei go-eCharger HomeFix 11 Boxen.
Bei einer funktioiniert der Download Link (Wert von "dll") für die Ladehistorie seit mehreren Tagen nicht - http status code 504 - Gateway Timeout. Für die andere Box funktioniert der Link.

Das ist jetzt nicht das erste mal, dass wir das Problem mit diesem Link haben...

Ich weiß, dass die offizielle Support-Seite über https://go-e.co/support-charger-home/ läuft. Aber für den Fall ist die Problembeschreibung und die erforderliche Kaufdokumentation (Datum, Händler, Kundenadresse wg. möglicher Rücksendung) einfach nur Overkill.

Den nicht funktioinierenden dll-Link schicke ich gerne per E-Mail oder PN, nicht aber öffentlich

Ventusfahrer, 2022-05-02:
Problem tritt aktuell wieder auf

naerschhersch, 2022-05-17:
Probier mal den telefonischen Kontakt, vermutlich kann dir so schon geholfen werden.