goecharger/go-eCharger-API-v2 #78: OCPP in API response

ianmurrays, 2022-04-26:
Hi,
I can see there are OCPP keys in the API response (I'm on 053.3 beta for what it's worth). Does this mean the charger now supports OCPP?

Thanks!

0xFEEDC0DE64, 2022-04-27:
Yes, ocpp is already implemented on the chargers for quite some time and at the moment i'm improving compatibility even more with every software update as we are already testing with multiple ocpp server solutions.

If you are interested you could join the go-e ocpp beta testers program, you can get a documentation on how to enable and verify the ocpp connection with the http-api.

Just contact the support and ask for the ocpp documentation pdf :)

ianmurrays, 2022-04-27:
Awesome thanks!

ianmurrays, 2022-04-27:
I contacted them via the form on the website, I hope that's what you referred to 🙂

closed 2022-10-17 by 0xFEEDC0DE64