goecharger/go-eCharger-API-v2 #81: Per trx Ladevorgang freischalten auf bestimmte CardId

kelvin1001, 2022-05-12:
Hallo!

Ich habe den - wirklich sehr aktiven - goe Support schon massiv genervt...komme aber leider nicht weiter. 

Ich möchte den Ladevorgang per API Key Aufruf Authentifizieren. 

Hierfür sende ich z.B. 
http://192.xxx.xxx.127/api/set?trx=1

Zurück erhalte ich 
{"trx":"trx: failed: starting transaction for 1 is not allowed (1)"}

Wenn ich null oder 0 sende, klappt es wie in der API Docu beschrieben (null = Authentifizierung erforderlich, 0 = keine Karte notwendig).

Wenn ich mit der Karte am Charger authentifiziere, springt der trx wie erwartet auf 1 um.

Jemand eine Idee?

Danke!!!

VG kelvin

Danny-clever-pv, 2022-06-04:
Hallo,
ich habe ein ähnliches Problem. Ich kann via cloud API die Variable trx gar nicht beschreiben.
Mache ich was falsch oder ist das ein bekanntes Problem?

Danke!
Danny

cribskip, 2022-08-29:
Hi,

ich würde auch sehr gern per API "wie ein Chip" freischalten, um die Verbräuche besser zuordnen zu können.
Vielleicht kann @peterpoetzi hier einmal schauen - das wäre sehr lieb und toll!

Viele Grüße
Sascha

moejetz, 2022-11-10:
Hi,

any news on this one?
I'd like to authenticate via API for a specific CardID as well - preferably over MQTT.
I bought this Wallbox especially because of the open API without the need to go over a cloud - and now the most important API call - authenticate and charge - doesn't work. This is so frustrating :(

I'm able to set `trx` to `1` via MQTT, but that's actually has no effect at all.
Also, if I authenticate and start charging via app, the `trx` value doesn't change at all in MQTT...

Any advice?

Thanks in advance,
Moritz

seh-len, 2022-11-19:
Hi,
im gleichen Zug dann bitte auch erlauben, dass man trx nachträglich auch wieder auf null setzen darf, ohne dass das Kabel abgesteckt wird.

> {'trx': 'trx: failed: cannot manually stop transaction when it has already been used to charge (null)'}

Danke!

Viele Grüße
Simon

4homeassistant, 2023-01-31:
> Hi,
> 
> ich würde auch sehr gern per API "wie ein Chip" freischalten, um die Verbräuche besser zuordnen zu können. Vielleicht kann @peterpoetzi hier einmal schauen - das wäre sehr lieb und toll!
> 
> Viele Grüße Sascha

Das wäre fein und ist eigentlich auch so in der Doku https://github.com/goecharger/go-eCharger-API-v2/blob/main/apikeys-de.md beschrieben. Leider klappt das Schreiben bei unseren beiden Ladern auch nicht.

moejetz, 2023-01-31:
FYI: Ich hatte das selbe Problem.
Mich interessieren die Karten an sich nicht, ich wollte nur eine Zugangskontrolle, die die Wallbox freischaltet wenn wir zuhause sind und sperrt wenn wir nicht zuhause sind.
Das habe ich folgendermaßen umgesetzt:
* Laden erlaubt: `acs=0` und `frc=2`
* Laden verboten: `acs=1` und `frc=1`

Die `trx` prop setze ich gar nicht.
Damit wird mein Auto automatisch geladen wenn es verbunden ist und ich zuhause bin, und sobald ich nicht zuhause bin wird kein Auto geladen.

Eine Sache die mir aufgefallen ist: die MQTT API hört nach ein paar Tagen einfach auf zu funktionieren bei mir.
Mein Abhilfe: ich habe die HTTP API aktiviert und mache jeden morgen um 5 einen Reboot der Wallbox. Seitdem passiert das nicht mehr. Der Reboot Command lautet: `http://<wallbox-ip>/api/set?rst=1`.

premultiply, 2023-01-31:
Vorsicht, hier werden jetzt zwei Wunschverhalten vermischt:

- RFID-Authentifizierung
- Zwangsfreigabe trotz fehlender Karte

moejetz, 2023-01-31:
> Vorsicht, hier werden jetzt zwei Wunschverhalten vermischt:
> 
> * RFID-Authentifizierung
> * Zwangsfreigabe trotz fehlender Karte

Stimmt, deshalb habe ich auch direkt dazu geschrieben dass ich mein Problem anders gelöst habe.
Natürlich hilft meine Lösung nichts wenn man die RFID-Authentifizierung automatisieren möchte.
Trotzdem ist das eventuell eine gangbare Alternative für manche Menschen weshalb ich das erwähnen wollte.

marcelgsteiger, 2023-04-19:
Es würde mir auch sehr helfen, eine CardID mit trx freischalten zu können. Der Grund liegt eigentlich an einem anderen "Fehler":  Auch wenn ein Fahrzeug permanent eingesteckt bleibt, geht eine anfangs gemachte RFID-Authentication je nach Fahrzeugtyp etwa mal verloren (nach Phasenumschaltungen oder Pause/Wiederaufnahme). Das könnte man via API problemlos auffangen, aber ich erhalte beim Setzen von trx immer die Fehlermeldung "failed: Starting transaction for 2 is not allowed (2)".
Am besten wäre es, die RFID-Authentication würde stabil funktionieren und immer erhalten bleiben, solange ein Fahrzeug nicht ausgesteckt wird.
Ich benutze verschiedene RFID-Chips für die Auswahl verschiedener Ladeprofile (z.B. "nur solar" oder "nachts fertigladen").

0xFEEDC0DE64, 2023-04-20:
funktioniert trx=1 in deinem fall nicht? geht die transaction verloren, ohne dass das auto abgesteckt wird?

marcelgsteiger, 2023-04-21:
ja, genau. Das tritt aber nur mit einem von vier Fahrzeugen auf, einem cupra (soll baugleich mit id.3 sein, aber mit einem id.3 eines anderen Parkplatzes an derselben Steuerung hatte ich dieses Problem nie). 
Ich versuche jetzt folgenden Workaround:
- ich speichere den Zeitstempel jedes Statuswechsels, insbes. auf Status 1 (kein Fahrzeug)
- wenn ich "authentication required" in Status 3 erhalte und der letzte Statuswechsel auf 2 (Fahrzeug lädt) nach dem letzten Wechsel auf Status 1 liegt, weiss ich, dass go-e ohne Ausstecken die transaction verloren hat und ich nehme die Ladung mit frc=2&acs=0 wieder auf
- wenn die Ladung beendet wurde aktiviere ich mit acs=1 den RFID-Chipzwang wieder (was nicht ausschliesst, dass sie wie oben wiederaufgenommen werden könnte)

Dabei verliere ich zwar die Zuordnung des gesamten Ladevorgangs zum "richtigen" RFID-Zähler, aber das ist hier das kleinere Übel (der betroffene Besitzer hat keinen Bedarf für chipgetrennte Zählung).
Da ich in der Textdarstellung des Chips ("Namenliste") auch das Ladeprofil codieren lasse (z.B. wird mit Voranstellen eines Sternchens beim Namen "reine Solarladung" verlangt), speichere ich auch das Profil des beim ersten Ladevorgang verwendeten Chips, so dass ich nach der Ausserbetriebnahme der Authentication mit acs=0 mit dem korrekten Ladeprofil weiterladen kann.
Im übrigen bin ich mit der offenen API-Ansteuerbarkeit des go-e sehr glücklich in diesen Zeiten hoher Energietarife.

4homeassistant, 2023-05-04:
> funktioniert trx=1 in deinem fall nicht? geht die transaction verloren, ohne dass das auto abgesteckt wird?

Das kann ich mit der 055.6 Beta bestätigen @0xFEEDC0DE64. Wenn ich das Fahrzeug anstecke (Ioniq 5) und trx wähle geht die Einstellung nach demselben Timeout verloren als wenn kein Fahrzeug verbunden ist. Ich teste das nochmal mit anderen Fahrzeugen aber bis zur 055.5 hat's noch funktioniert.