goecharger/go-eCharger-API-v2 #84: MQTT Topic beginnt mit /

digidax, 2022-08-01:
Wäre es möglich, dass die MQTT Topic NICHT mit / beginnt, sondern direkt mit go-eCharger/123456

Ist im MQTT Explorer schöner go-eCharger gleich in der Root Topic zu sehen anstatt ein eeres Feld:
[image]
[image]

0xFEEDC0DE64, 2022-08-04:
als kurzfristigen work-around:
api key "mtp", type string, "mqtt topic prefix, set to null to reset to default again"

copacetic82, 2022-09-27:
Hat sich dieses Verhalten in der neusten version 054.6 Beta geädnert, ohne es in den release notes erwöhnt wurde? Bei mir waren die topics nach dem upgrade auf einer einer anderen ebene, und diverse sensoren funktionierten dadurch nicht. Hab vorerst ein rollback auf 054.3 gemacht. Wird das auch in die nächste non-beta version übernommen? 

Siehe vergleich: 
[image]

arboeh, 2022-09-27:
Kann ich bestätigen und ich habe mir in den letzten Stunden nen Wolf gesucht, weil ich heute eine neue Instanz aufgesetzt habe... :-|

copacetic82, 2022-09-27:
> Kann ich bestätigen und ich habe mir in den letzten Stunden nen Wolf gesucht, weil ich heute eine neue Instanz aufgesetzt habe... :-|

Ging mir ähnlich, konnte es aber zum glück recht schnell eingrenzen. Wäre cool gewesen wenns in den release notes stehen würde :/

syssi, 2022-09-28:
Same here:

```
-/go-eCharger/072246/fhz 49.754
+go-eCharger/072246/fhz 49.754
```

copacetic82, 2022-10-01:
fixed in 054.7 Beta

digidax, 2022-10-01:
The forward slash (/) is used to separate each level within a topic tree.
Focus is the separation not the beginning:
[http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718106](url)

MQTT Best practices
Never use a leading forward slash

A leading forward slash is permitted in MQTT. For example, /myhome/groundfloor/livingroom. However, the leading forward slash introduces an unnecessary topic level with a zero character at the front. The zero does not provide any benefit and often leads to confusion.
[https://www.hivemq.com/blog/mqtt-essentials-part-5-mqtt-topics-best-practices/](url)

fu-zhou, 2022-10-09:
in der 54.7 hat sich irgendwie auch verändert, wie der go-e charger seine Datenpunkte über MQTT veröffentlicht - s. https://github.com/goecharger/go-eCharger-API-v2/issues/89#issuecomment-1272274910

0xFEEDC0DE64, 2022-11-03:
ich wollte eigentlich nur für neue mqtt-installationen endlich den / zu beginn los werden, in der ersten beta hat's aber den default auch für bestehende mqtt-user mit verdreht, mit dem nächsten beta update sollte es jetzt so sein, dass wenn mqtt schon aufgedreht war, er den führenden / behält und für neue mqtt user per default nur ohne / vorschlägt und man kann manuell in der app den mqtt topic prefix selber nach wunsch verändern.

erxbout, 2024-12-17:
duplicate of https://github.com/goecharger/go-eCharger-API-v2/issues/44

Ich schließe dieses issue, da das topic in der app inzwischen einstellbar ist und der default ohne leading "/" kommt

closed 2024-12-17 by erxbout