goecharger/go-eCharger-API-v2 #87: WebSocket 0.54 API sends defective packets

chk2902, 2022-08-25:
This is what I get after authentication. 

I guess sending the fullStatus packets does cause timing issues and concatenates (even once by truncating the fullStatus packet) the delta packets:

` {"type":"fullStatus","partial":true,"status":{"mod":1,"rfb":1646,"stao":null,"alw":true,"acu":6,"acui":6,"adi":true,"acl":16,"dwo":null,"tpa":0,"sse":"056129","eto":472715,"etop":472715,"wifis":[...],"scas":2}}`

`{"type":"fullStatus","partial":true,"status":{"scan":[...],"lwf":null,"scaa":4270,"wst":3,"wsc":0,"wsm":"","wsms":3,"ccw":{"ssid":"chk-h10-g","encryptionType":3,"pairwiseCipher":4,"groupCipher":4,"b":true,"g":true,"n":true,"lr":false,"wps":false,"ftmResponder":false,"ftmInitiator":false,"channel":12,"bssid":"...","ip":"192.168.2.59","netmask":"255.255.255.0","gw":"192.168.2.1","ipv6":["..."],"dns0":"192.168.2.1","dns1":"0.0.0.0","dns2":"0.0.0.0"},"wfb":null,"wcb":"...","wpb":null,"nif":"st","dns":{"dns":"0.0.0.0"},"hsts":"go-echarger_0xxxxx","hsta":"go-echarger_0xxxxx","host":"go-echarger_0xxxxx","rssi":-37,"lssfc":null,"lsstc":7640,"ehs":288408,"efh":101376,"efh32":101376,"efh8":63048,"fem":1,"sbe":true,"emfh":65084,"emhb":47092}}`

`{"type":"fullStatus","partial":true,"status":{"eci":{"model":1,"features":50,"cores":2,"revision":3},"ecf":{"source":1,"source_freq_mhz":320,"div":2,"freq_mhz":160},"efi":null,"abm":"C4:5B:BE:80:4B:40","dbm":"C4:5B:BE:80:4B:40","cbm":null,"obm":null,"macs":"C4:5B:BE:80:4B:40","maca":"C4:5B:BE:80:4B:41","esr":[12,12],"rr":3,"wae":true,"fwan":"go-e-056129","wan":"go-e-056129","facwak":true,"wak":false,"wda":false,"wen":true,"wsmr":-90,"wapc":1,"waap":3,"swc":false,"tse":false,"ts":"europe.pool.ntp.org","tssm":0,"tssi":3600000,"tsss":0,"tsom":0,"tof":60,"tds":1,"utc":"2022-08-25T05:28:38.024.060","loc":"2022-08-25T07:28:38.024.876 +02:00","led":{"id":18,"name":"Ready","norwayOverlay":true,"modeOverlay":true,"subtype":"renderCmds","ranges":[{"from":0,"to":5,"colors":["#0000FF"]},{"from":6,"to":31,"colors":["#000000"]}]},"ledo":null,"lbr":144,"lmo":3,"ama":16,"la1":16,"la3":16,"clp":[6,10,12,14,16],"al1":6,"al2":10,"al3":12,"al4":14,"al5":16,"bac":1,"sdp":2,"lbp":null,"amp":6,"ffna":"go-eCharger_056129","fna":"go-eCharger_056129","cy":"un","mukw":null,"cid":"#0000FF","cwc":"#FFFF00","cch":"#00FFFF","cfi":"#00FF00","ust":2,"lck":2,"sch_week":{"control":0,"ranges":[{"begin":{"hour":0,"minute":0},"end":{"hour":23,"minute":0}},{"begin":{"hour":23,"minute":0},"end":{"hour":0,"minute":50}}]},"sch_satur":{"control":0,"ranges":[{"begin":{"hour":0,"minute":0},"end":{"hour":0,"minute":0}},{"begin":{"hour":0,"minute":0},"end":{"hour":0,"minute":0}}]},"sch_sund":{"control":0,"ranges":[{"begin":{"hour":0,"minute":0},"end":{"hour":0,"minute":0}},{"begin":{"hour":0,"minute":0},"end":{"hour":0,"minute":0}}]},"nmo":false,"dco":true,"fsp":true,"frci":true,"acs":0,"frc":0,"rbc":82,"rbt":265414,"car":1,"err":0,"cbl":32,"pha":[false,false,false,true,true,true],"wh":0,"trx":null,"fwv":"054.0","arv":"1.2.1","ccu":null,"oem":"go-e","typ":"go-eCharger","fwc":"11","ccrv":"11","apd":{"project_name":"goecharger_hw2+","version":"054.0","secure_version":0,"timestamp":"Aug 10 2022 14:28:19","idf_ver":"e51d85039","sha256":"89a749f0a8929c19"},"otap":{"type":0,"subtype":16,"address":1703936,"size":4194304,"label":"app_0","encrypted":true}}}`

`{"type":"fullStatus","partial":true,"status":{"part":[{"type":1,"subtype":2,"address":65536,"size":262144,"label":"nvs","encrypted":false},{"type":1,"subtype":2,"address":327680,"size":262144,"label":"nvsFac","encrypted":false},{"type":1,"subtype":1,"address":589824,"size":4096,"label":"phy_init","encrypted":false},{"type":1,"subtype":0,"address":593920,"size":8192,"label":"otadata","encrypted":true},{"type":0,"subtype":0,"address":655360,"size":1048576,"label":"factory","encrypted":true},{"type":0,"subtype":16,"address":1703936,"size":4194304,"label":"app_0","encrypted":true},{"type":0,"subtype":17,"address":5898240,"size":4194304,"label":"app_1","encrypted":true},{"type":1,"subtype":4,"address":10092544,"size":4096,"label":"nvs_key","encrypted":true},{"type":1,"subtype":3,"address":10096640,"size":65536,"label":"coredump","encrypted":true},{"type":64,"subtype":1,"address":10162176,"size":65536,"label":"keys","encrypted":true},{"type":1,"subtype":130,"address":10227712,"size":4194304,"label":"spiffs","encrypted":false}],"pto":61440,"lse":true,"cdi":{"type":1,"value":0},"lch":null,"lccfi":null,"lccfc":null,"lcctc":null,"tma":[29,35.125],"amt":32,"nrg":[233,232,232,0,0,0,0,0,0,0,0,0,0,0,0,0],"modelStatus":15,"msi":15,"lmsc":36115,"mca":6,"awc":0,"awp":3,"awcp":{"start":1661403600,"end":1661407200,"marketprice":66.824},"awpl":[{"start":1661403600,"end":1661407200,"marketprice":66.824},{"start":1661407200,"end":1661410800,"marketprice":69},{"start":1661410800,"end":1661414400,"marketprice":64.02},{"start":1661414400,"end":1661418000,"marketprice":61.386},{"start":1661418000,"end":1661421600,"marketprice":61.622},{"start":1661421600,"end":1661425200,"marketprice":62.84},{"start":1661425200,"end":1661428800,"marketprice":63.386},{"start":1661428800,"end":1661432400,"marketprice":63.516},{"start":1661432400,"end":1661436000,"marketprice":61.982},{"start":1661436000,"end":1661439600,"marketprice":64.5},{"start":1661439600,"end":1661443200,"marketprice":68.68},{"start":1661443200,"end":1661446800,"marketprice":71.423},{"start":1661446800,"end":1661450400,"marketprice":76.679},{"start":1661450400,"end":1661454000,"marketprice":76},{"start":1661454000,"end":1661457600,"marketprice":71},{"start":1661457600,"end":1661461200,"marketprice":67.34},{"start":1661461200,"end":1661464800,"marketprice":62.673}],"ido":null,"frm":1,"fup":true,"awe":false,"fst":1400,"fmt":300000,"att":21600,"ate":0,"ara":false,"acp":true,"cco":18,"esk":true,"fzf":false,"sh":200,"psh":500,"po":-300,"zfo":200,"psmd":0,"sumd":0,"mpwst":120000,"mptwt":600000,"ferm":1,"mmp":0,"tlf":false,"tls":null,"atp":null,"lpsc":null,"inva":null,"pgrid":null,"ppv":null,"pakku":null,"deltap":0,"pnp":0,"deltaa":0,"pvopt_averagePGrid":0,"pvopt_averagePPv":0,"pvopt_averagePAkku":0,"ct":"renaultZoe","mci":0,"mcpd":0,"mcpea":null,"su":true,"sua":false,"hws":true,"hsa":true,"var":11,"loe":false,"log":"","lop":50,"lof":0,"map":[1,2,3],"upo":false,"pwm":0,"lfspt":null,"fsptws":null,"spl3":4200,"psm":1}}`

`{"type":"fullStatus","partial":true,"status":{"ocu":["054.0","V 053.0 OUTDATED","V 053.1 OUTDATED","V 053.3 OUTDATED"],"ocuca":true,"ocs":0,"ocp":0,"ocm":"","ocl":null,"oc�~���{"type":"fullStatus","partial":false,"status":{"cms":"_go-e_controller","cmp":"_tcp","lcs":0,"csa":false,"ctrls":null,"ccd":null}}�~�{"type":"deltaStatus","status":{"rfb":1649,"acu":6,"acui":6,"rssi":-36,"ehs":288056,"efh":84688,"efh32":84688,"efh8":46360,"emhb":31732,"utc":"2022-08-25T05:28:38.361.480","loc":"2022-08-25T07:28:38.362.494 +02:00","led":{"id":18,"name":"Ready","norwayOverlay":true,"modeOverlay":true,"subtype":"renderCmds","ranges":[{"from":0,"to":5,"colors":["#0000FF"]},{"from":6,"to":31,"colors":["#000000"]}]},"rbt":265751,"car":1,"err":0,"pha":[false,false,false,true,true,true],"ccu":null,"fwc":"11","tma":[29,35.125],"nrg":[233,232,232,0,0,0,0,0,0,0,0,0,0,0,0,0],"modelStatus":15,"msi":15,"lmsc":36115,"lfspt":null,"cpe":true,"fhz":49.904,"rcd":4,"wcch":1,"qsc":0,"qsw":3,"lps":61,"tpcm":[14,0,3,2,0,0,0,0,35,0,3,0,0,61,28,0,0,0,0,0,0,0,0,0,0,4,0],"data":"{\"i\":120,\"url\":\"https://data.v3.go-e.io/data_ng?e=Bpy8zEu0XMCDaOta_FC32oEEuzTmSvor4QlnQYTXTmC1PUhUluN-7t4ZzrYz8v4GIw2uTAGi4cv4VR47XNV5asuFFRdom_I\"}","hlasl":17,"mscs":3}}�~Y{"type":"deltaStatus","status":{"rfb":1647,"ehs":288152,"efh":94640,"efh32":94640,"efh8":56312,"utc":"2022-08-25T05:28:39.362.134","loc":"2022-08-25T07:28:39.363.144 +02:00","rbt":266751,"nrg":[234,232,232,0,0,0,0,0,0,0,0,0,0,0,0,0],"fhz":49.875,"rcd":null,"qsw":2,"lps":48,"tpcm":[15,0,1,1,0,0,0,0,31,0,3,0,65,61,27,0,100,0,0,0,0,0,0,0,0,6,0]}}�~o{"type":"deltaStatus","status":{"rfb":1649,"rssi":-38,"ehs":288568,"efh":106064,"efh32":106064,"efh8":67736,"emhb":42996,"utc":"2022-08-25T05:28:40.340.724","loc":"2022-08-25T07:28:40.341.834 +02:00","rbt":267729,"nrg":[233,233,232,0,0,0,0,0,0,0,0,0,0,0,0,0],"fhz":49.796,"rcd":4,"qsw":0,"lps":54,"tpcm":[15,0,1,3,0,0,0,0,36,1,53,0,65,62,26,0,0,0,0,0,0,0,0,0,0,4,0]}}�~h{"type":"deltaStatus","status":{"rfb":1652,"rssi":-37,"ehs":288616,"efh":106436,"efh32":106436,"efh8":68108,"emhb":46068,"utc":"2022-08-25T05:28:41.363.583","loc":"2022-08-25T07:28:41.364.613 +02:00","rbt":268752,"nrg":[233,233,231,0,0,0,0,0,0,0,0,0,0,0,0,0],"fhz":49.887,"wcch":0,"lps":58,"tpcm":[17,0,1,1,0,0,0,0,35,0,28,0,65,62,28,0,0,0,0,2,0,0,0,0,0,4,0]}}�~\{"type":"deltaStatus","status":{"rfb":1650,"rssi":-38,"ehs":288648,"efh":108032,"efh32":108032,"efh8":69704,"emhb":47092,"utc":"2022-08-25T05:28:42.339.553","loc":"2022-08-25T07:28:42.340.562 +02:00","rbt":269728,"nrg":[234,232,232,0,0,0,0,0,0,0,0,0,0,0,0,0],"rcd":null,"lps":56,"tpcm":[17,0,2,1,0,0,0,0,31,0,3,0,65,61,28,0,0,0,0,0,0,1,0,0,0,4,0]}}�~B`

chk2902, 2022-08-25:
... and the second connect often works, after login the data appears "normal".

chk2902, 2022-10-05:
seems to be fixed (054.7)

erxbout, 2024-07-10:
I will close the issue then

closed 2024-07-10 by erxbout