goecharger/go-eCharger-API-v2 #88: Firmware 054 regression: cannot turn the station on/off with frc

erikderzweite, 2022-09-02:
Since Firmware 054 I cannot start or stop charging process with frc api flag. 
I use MQTT and python, set /go-eCharger/charging_station_number/frc/set to 1 and 2. 053 works as expected 054 does not.

Emacee, 2023-08-12:
Same for me, the cable doesn't unlock when forced to Neutral (0) or Off (1). The charging itself however is paused. This is based on the latest beta firmware 0.55.7
Would be helpful if this frc command would also force an unlock of the cable to end the charging session.

0xFEEDC0DE64, 2023-10-25:
there is an api key to define the cable unlock mode.
you have to set the value so that it unlocks everytime there is a charge pause

please try setting:

ust=1