goecharger/go-eCharger-API-v2 #89: ab Firmware 54.x läuft MQTT nicht mehr

Schwunkel, 2022-09-03:
nach dem Update von 53.3 auf 54.0 bzw. 54.1 läuft MQTT nicht mehr. Alle paar Sekunden kommt ein neues publish und subscribe (müllt das Server-log zu) aber es kommt keine richtige Verbindung zustande. Nach dem Downgrade auf 53.3 funktioniert es wieder.

Grüße
Schwunkel

PS: der MQTT-Server läuft bei mir unter ioBroker

fu-zhou, 2022-09-04:
Hallo Schwunkel, ich habe eben auf 54.0 upgegraded und MQTT mit iobroker als Server läuft bei mir nach wie vor einwandfrei.... Hast du mal den Adapter gestoppt, die goecharger Objekte unter MQTT gelöscht und den Adapter wieder gestartet? Da werde die MQTT Objekte dann neu angelegt - nur so eine Idee...

Schwunkel, 2022-09-05:
nach vielen Up-/Downgrades, Neustarts, Objekte löschen,... habe ich es eingegrenzt: der Fehler tritt ab und an dann auf, wenn bei aktiver Verbindung der Server neugestartet wird (ioBroker Neustart, Adapter Neustart ist anscheinend kein Problem).
Betrifft nur go-e ab FW 54.x, andere Clients laufen problemlos.

kolossboss, 2022-09-28:
Mit 054.6 läuft bei mir mqtt auch nicht mehr.
Habt downgraded auf 054.3 und alles geht wieder.

copacetic82, 2022-10-01:
verhalten in 054.7 Beta wieder wie gewohnt

fu-zhou, 2022-10-08:
> verhalten in 054.7 Beta wieder wie gewohnt

Leider nein - ich nutzte auch den MQTT Adapter in iobroker und steuere z.B. Ladestrom und Anzahl Phasen über ein blockly Skript (amp + psm). Bei der 54.7 beta stehen alle Werte im "root" und die Werte können nicht mehr über das Skript geschrieben werden, weil das Unter-Objekt "set" in den schreibbaren Objekten fehlt:

54.7 beta Objektbaum:
[image]

54.3 Objektbaum:
[image]