goecharger/go-eCharger-API-v2 #90: go-e config failed

papagooo, 2022-09-11:
[image]

After filling hostname and serial got an error like 
User input malformed: expected float for dictionary value @ data['correction_factor']

0xFEEDC0DE64, 2022-09-12:
homeassistant once explicity rejected our go-e implementation for go-eChargers, but you found anotehr one? The setup dialog you see is not developed by us (maybe @cathiele).

Also "correction_factor" is no api key that I know about, the closest we have is the power factor which is part of the api key nrg, which is unchanged since many many years.

cathiele, 2022-10-02:
it was a bug in my integration that prevented the setup flow. please check version 0.52.2 this should fix it. my old Charger (V2-Hardware) is way of in the power values. to this factor corrects the values. its generally not necessary to set this other than 1.0 if you don't care or have a charger with exact readings.

cathiele, 2022-10-02:
> homeassistant once explicity rejected our go-e implementation for go-eChargers, but you found anotehr one? The setup dialog you see is not developed by us (maybe @cathiele).

my integration is available via the HACS-Integration. I just did it because there was no official integration. @0xFEEDC0DE64  do you know why it got rejected?

erxbout, 2024-06-26:
We support HA autodiscovery via mqtt nowadays.. (hardware_v3+)
I will close this issue as is an issue of https://github.com/cathiele/homeassistant-goecharger
Please add a hint that this integration is using the http api_v1

closed 2024-06-26 by erxbout