goecharger/go-eCharger-API-v2 #91: Autocharge Support?

dabeck, 2022-09-13:
Kann der go-e Charger das Auto via EVCC ID identifizieren und so den Ladevorgang Authentifizieren?

Ich habe in diesem [Paper](https://cdn.vector.com/cms/content/products/vcharm/docs/WhitePaper_Identification_of_ElectricVehicles_in_CSMS_via_OCPP.pdf) folgendes gelesen:

> Autocharge: The charging station uses the EVCC ID (MAC address of EVCC) for authorization of the vehicle.

4homeassistant, 2023-01-31:
@0xFEEDC0DE64, hast Du eine Idee, wie EnBW das mit Autocharge umsetzt? Deren Verfahren arbeitet ja vermutlich nicht innerhalb der ISO15118.

dabeck, 2023-01-31:
@4homeassistant Leider findet über AC-Kabel überhaupt keine aktive Kommunikation statt. Ich habe es mittlerweile so gelöst, dass ich einen Service entwickelt habe der via MQTT die Signale vom Go-E abfängt und sobald ein Auto angeschlossen wurde die Tesla-API fragt ob es mein Auto ist. ;) (Standort, Ladekabel verbunden etc.)

4homeassistant, 2023-01-31:
Danke Dir, der Denkansatz ist gut. Ich vermute, dass das bei uns mit drei unterschiedlichen Fabrikaten eine Herausforderung wird, die den Nutzen übersteigt. Aber immer wieder gut, andere Ideen kennenzulernen 😊