goecharger/go-eCharger-API-v2 #92: Feature request: AMP stufenlos regeln?

ako-3004, 2022-09-17:
Bisher dachte ich die Ampére-Vorgabe, über das pwm wäre auf ganzzahlige Vorgaben definiert, also auch Euer "amp".
Ich hab jetzt bei Freunden charger kennengelernt, die die Leistung aber stufenlos an eine Vorgabe (PV Überschuss) anpassen.
Mein PV Überschussladen basiert auch auf ganzzahligen amp. 

Ist es möglich oder habt ihr geplant die Leistungsvorgabe stufenlos regeln zu können?
Falls ja, wie würde das funktionieren wenn nicht über den amp Wert?

jw-rav, 2022-09-23:
Hierzu gibt es bereits das Ticket #63 

Leider bislang keine Rückmeldung, ob in dieser Richtung etwas getan wird.

0xFEEDC0DE64, 2023-01-09:
kommt irgendwann.
ist aber noch schwer abzuschätzen, wann genau, da der Aufwand beim Implementieren und dem anschließenden Testen sehr hoch ist und wir aktuell dringendere Themen haben.

fre4242, 2026-03-06:
So langsam gebe ich die Hoffnung auf