goecharger/go-eCharger-API-v2 #99: 054.7 Final: MQTT Objekte haben kein "set" mehr

fu-zhou, 2022-10-17:
Ich nutzte auch den MQTT Adapter in iobroker und steuere z.B. Ladestrom und Anzahl Phasen über ein blockly Skript (amp + psm). Bei der 54.7 stehen alle Werte im "root" und die Werte können nicht mehr über das Skript geschrieben werden, weil das Unter-Objekt "set" in den schreibbaren Objekten fehlt:

54.7 Objektbaum:
[image]

54.3 Objektbaum:
[image]

_Originally posted by @fu-zhou in https://github.com/goecharger/go-eCharger-API-v2/issues/89#issuecomment-1272274910_

0xFEEDC0DE64, 2022-11-03:
In einer der letzten Versionen habe ich eingezogen, dass der Charger auch endlich auf ungültige Schreib-Anfragen antwortet.
Der Charger subscribed jetzt nicht mehr für 100 api keys einzeln (was kurzzeitig sehr viel RAM frisst), sondern subscribed nur mehr einmal auf ein wildcard topic.

Ist der mqtt server im iobroker großartig anders als der von mosquitto? Lässt sich der gescreenshottete test irgendwie nachstellen?

[image]

fu-zhou, 2022-11-03:
Da bin ich zu sehr Anwender, als Entwickler... Der iobroker mqtt Server schreibt die Werte von den Clients in die Objektdatenbank (s. Screenshots oben), diese Objekte werden dann in den Skripten adressiert. Ein Terminal-Fenster gibt es da nicht und auch nicht die Möglichkeit (für mich mit meinen Fähigkeiten), die Objekte mit einem Parameter zu ergänzen (z.B. amp**/set**). 

Was muss denn der Server machen/ können, um bei der aktuellen Firmware wieder den vollen Umfang einzulesen? Vielleicht kann ich das dann beim iobroker mqtt Adapter Entwickler hier in Github https://github.com/ioBroker/ioBroker.mqtt einspeisen?

Der Konfigurationsdialog vom Adapter sieht so aus:
[image]
Erkennst du da eine Option, die das macht, was ich brauche? Ich habe schon ein bischen rumprobiert, bin aber nicht zum gewünschten Ergebnis gekommen

tops4u, 2022-11-19:
Ich habe das gleiche Problem. 

Beim Connect sieht es so aus, als würde die Subscription funktionieren:

`Client [go-echarger_xxxxxx] subscribes on "go-eCharger/xxxxxx/+/set" with regex /mqtt\.0\.go-eCharger\.xxxxxx\.[^.]*\.set/`
 `Client [go-echarger_xxxxxx] subscribes on "go-eCharger/xxxxxx/+/set" with regex /go-eCharger\.xxxxxx\.[^.]*\.set/`

Wenn ich dann allerdings in IOBroker das Set Objekt ändere (Objekt ID: mqtt.0.go-eCharger.xxxxxx.amp.set) dann ändert sich da gar nichts mehr. 

Ich denke der Charger verwendet diese Regex für die Subscription: _go-eCharger\.xxxxxx\.[^.]*\.set/_
Die zieht allerdings beim MQTT Adpater auf IOBroker nicht. Diese hier würde gehen: _go-eCharger\.xxxxxx\.[a-zA-Z0-9]*\.set_

tops4u, 2022-11-19:
Abgesehen davon zeigt mir die Go.E App die Changelogs für 54.8 mit Bugfixes für MQTT an, jedoch steht oben nur bis 54.7 als Installierbar zur Auswahl... 

Grundsätzlich würde ich lieber Modbus TCP verwenden, da gibt es jedoch bisher keine Möglichkeiten die Phasen Anzahl zu schalten.

fu-zhou, 2022-11-19:
> Abgesehen davon zeigt mir die Go.E App die Changelogs für 54.8 mit Bugfixes für MQTT an, jedoch steht oben nur bis 54.7 als Installierbar zur Auswahl...
> 
> Grundsätzlich würde ich lieber Modbus TCP verwenden, da gibt es jedoch bisher keine Möglichkeiten die Phasen Anzahl zu schalten.

Ist bei mir auch so, habe gedacht, dass sich die 54.8 vielleicht versteckt und mal das Update laufen lassen - dann wird dir 54.7 installiert. Warten wir mal ab...

tops4u, 2022-11-19:
> Warum würdest du Modbus TCP vorziehen?

Weil MQTT mir viel zu viel Zeugs liefert. Bei Modbus TCP habe ich besser Kontrolle über was ich haben will, wie es heissen soll und wie oft es aktualisiert wird. Allerdings ist die Konfig etwas "schmerzhafter". Da ich die Übung allerdings bereits für meine PV-Wechselrichter gemacht habe, weiss ich nun wie es geht.

fu-zhou, 2022-11-19:
Ist ein Argument. Hast du nen SolarEdge WR? Hättest du Lust, mal deine aktuelle iobroker Modbus-Adapter Konfiguration zu posten (für WR und für den goE Charger)? Oder hast du das schon mal irgendwo gezeigt, z.B. im iobroker Forum?

Danke vorab...

tops4u, 2022-11-25:
Mit der aktuellen Version (54.9) funktioniert bei mir wieder alles.

fu-zhou, 2022-11-25:
bei der 54.9 funktioniert es bei mir nur, wenn ich den Objekt-Baum beibehalte, wie er unter der 54.3 angelegt wurde (also "set" und "result" unterhalb von "amp"). Wenn ich den Objektbaum lösche und bei der ersten MQTT-Verbindung mit der 54.9 wieder anlegen lasse, habe ich das Problem wie oben beschrieben. Wie sieht das "funktioniert" unter 54.9 bei dir aus?

tops4u, 2022-11-25:
> bei der 54.9 funktioniert es bei mir nur, wenn ich den Objekt-Baum beibehalte, wie er unter der 54.3 angelegt wurde (also "set" und "result" unterhalb von "amp"). 

Ja das ist insofern zu erwarten, da GO.E nun eben nicht mehr eine Subscription für alle Objekte macht, sondern ein Wildcard. Somit kann der Broker die möglichen SET Objekte ja gar nicht kennen. Um also einen Wert zu schreiben, musst DU explizit ein SET Objekt unterhalb dem angezeigten Objekt anlegen in IOBroker.

fu-zhou, 2022-11-27:
Hmh... wie lege ich das an? Mit einem Script? Mit CreateState unter Blockly wird das Objekt allerdings in der Javascript-Instanz angelegt, nicht unter MQTT... Der Objektbaum lässt mich unter einem State keinen weiteren State mehr anlegen, obwohl "amp" in der 54.3 ein State ist (allerdings als Ordner dargestellt) mit den unterlagerten States "set" und "result".
Insofern wäre mit Modbus mittlerweile auch lieber, wenn die API schon voll umfänglich integriert wäre (Phasenumschaltung etc.)

fu-zhou, 2022-12-28:
Hab's mit dem ioBroker Forum gelöst: Auf das Objekt "go-eCharger" gehen, dort einen neuen Datenpunkt anlegen:
"number" amp.set
"string" amp.result

und schon passt's wieder...

closed 2022-12-28 by fu-zhou