goecharger/go-eCharger-API-v2 #105: OCPP issues

bkonar6, 2022-11-30:
Hi,

i am trying to set charging profile , it seems that the charger accept the profile but it does not really change the amperage (through http API is ok) . Also it seems that the charger does not send any Wh measurement during the heatbeat but only at the end of the transaction.

 [INFO ] 2022-11-30 15:40:32,896 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=petroupoli2, sessionId=6b7eec98-4a2d-00f5-ba00-ec66b59e4711] Sending: [2,"49729321-b556-4c5d-81d5-9255ca251a4d","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":7,"stackLevel":3,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Absolute","chargingSchedule":{"duration":86400,"startSchedule":"2022-11-30T15:37:00.000Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":10,"numberPhases":3}],"minChargingRate":10}}}]
[INFO ] 2022-11-30 15:40:33,473 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=petroupoli2, sessionId=6b7eec98-4a2d-00f5-ba00-ec66b59e4711] Received: [3,"49729321-b556-4c5d-81d5-9255ca251a4d",{"status":"Accepted"}]

bkonar6, 2022-12-23:
New 0.55 beta firmware feedback:

a.Also i think that when the charger gets a grid power supply outage should keep the transaction in memory and upon reboot should send a stop transaction.
SOLVED

b.Other issue is that 'MeterValues.req' is not received correctly by SteVe during the transaction period
SOLVED

c.i am trying to set charging profile , it seems that the charger accept the profile but it does not really change the amperage (through http API is ok) .

Pending or I am doing something wrong.

Danny-clever-pv, 2023-02-05:
We have the same problem, the charge doesn't change the charging current via "SetChargingProfile", do you have an approved request? What do you recommend for dynamic PV charging to avoid wearing out the memory?

0xFEEDC0DE64, 2023-02-20:
Wearing out memory isnt possible, if we still have this in the documentation, then this is a leftover from the previous generations chargers. Setting the current with ocpp charging profiles should work and has been tested by go-e staff and customers. How exactly doesnt it work? Does the charger not accept the new limit, or is it accepted but not applied to the charging logic?

Danny-clever-pv, 2023-02-20:
we send: [2,"682570","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":0,"chargingProfilePurpose":"ChargePointMaxProfile","chargingProfileKind":"Absolute","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16}],"startSchedule":"2023-02-20T16:13:48.8642486Z"}}}]
The charger does not accept the new limit.

0xFEEDC0DE64, 2023-02-21:
What does the charger respond to that request?

Danny-clever-pv, 2023-02-21:
rejected 🫤

knese, 2023-03-03:
Same here:
Trying to use evcc.io with OCPP against go-e Charger Genesis:
```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/03/03 09:21:02 evcc 0.114.0
[main  ] INFO 2023/03/03 09:21:02 using config file: /config/evcc.yaml
[main  ] INFO 2023/03/03 09:21:02 starting ui and api at :7070
[db    ] INFO 2023/03/03 09:21:02 using sqlite database: /data/evcc.db
[ocpp  ] DEBUG 2023/03/03 09:21:16 chargepoint connected: go-e
[ocpp  ] TRACE 2023/03/03 09:21:16 received JSON message from go-e: [2,"531531560","BootNotification",{"chargePointVendor":"go-e","chargePointModel":"go-eCharger_V4","chargePointSerialNumber":"20XXXX","firmwareVersion":"055.5"}]
[ocpp  ] TRACE 2023/03/03 09:21:16 handling incoming CALL [531531560, BootNotification] from go-e
[ocpp  ] TRACE 2023/03/03 09:21:16 sent CALL RESULT [531531560] for go-e
[ocpp  ] TRACE 2023/03/03 09:21:16 sent JSON message to go-e: [3,"531531560",{"currentTime":"2023-03-03T08:21:16Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2023/03/03 09:21:16 received JSON message from go-e: [2,"531531561","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2023-03-03T08:21:16.000Z"}]
[ocpp  ] TRACE 2023/03/03 09:21:16 handling incoming CALL [531531561, StatusNotification] from go-e
[ocpp  ] TRACE 2023/03/03 09:21:16 sent CALL RESULT [531531561] for go-e
[ocpp  ] TRACE 2023/03/03 09:21:16 sent JSON message to go-e: [3,"531531561",{}]
[ocpp  ] TRACE 2023/03/03 09:21:16 received JSON message from go-e: [2,"531531562","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2023-03-03T08:21:16.000Z"}]
[ocpp  ] TRACE 2023/03/03 09:21:16 handling incoming CALL [531531562, StatusNotification] from go-e
[ocpp  ] TRACE 2023/03/03 09:21:16 sent CALL RESULT [531531562] for go-e
[ocpp  ] TRACE 2023/03/03 09:21:16 sent JSON message to go-e: [3,"531531562",{}]
[lp-1  ] WARN 2023/03/03 09:21:16 phases not configured, assuming 3p
[site  ] INFO 2023/03/03 09:21:17 site config:
[site  ] INFO 2023/03/03 09:21:17   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/03/03 09:21:17     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/03/03 09:21:17     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/03/03 09:21:17   vehicles:
[site  ] INFO 2023/03/03 09:21:17     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/03/03 09:21:17 loadpoint 1:
[lp-1  ] INFO 2023/03/03 09:21:17   mode:        off
[lp-1  ] INFO 2023/03/03 09:21:17   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/03/03 09:21:17   meters:      charge ✗
[lp-1  ] INFO 2023/03/03 09:21:17 car connected
[lp-1  ] INFO 2023/03/03 09:21:17 start charging ->
[lp-1  ] INFO 2023/03/03 09:21:17 vehicle updated: unknown -> Schneeball
[lp-1  ] WARN 2023/03/03 09:21:17 charger logic error: disabled but charging
[ocpp  ] TRACE 2023/03/03 09:21:17 enqueued CALL [3487313002, RemoteStopTransaction] for go-e
[ocpp  ] TRACE 2023/03/03 09:21:17 dispatched request 3487313002 for go-e
[ocpp  ] TRACE 2023/03/03 09:21:17 sent JSON message to go-e: [2,"3487313002","RemoteStopTransaction",{"transactionId":0}]
[ocpp  ] TRACE 2023/03/03 09:21:17 started timeout timer for go-e
[ocpp  ] TRACE 2023/03/03 09:21:17 received JSON message from go-e: [3,"3487313002",{"status":"Rejected"}]
[ocpp  ] TRACE 2023/03/03 09:21:17 handling incoming CALL RESULT [3487313002] from go-e
[ocpp  ] TRACE 2023/03/03 09:21:17 completed request 3487313002 for go-e
[lp-1  ] ERROR 2023/03/03 09:21:17 charger: Rejected
[ocpp  ] TRACE 2023/03/03 09:21:17 go-e ready to transmit again
[ocpp  ] TRACE 2023/03/03 09:21:17 timeout canceled for go-e
[lp-1  ] WARN 2023/03/03 09:21:18 charger logic error: disabled but charging
[ocpp  ] TRACE 2023/03/03 09:21:18 enqueued CALL [1200002614, RemoteStopTransaction] for go-e
[ocpp  ] TRACE 2023/03/03 09:21:18 dispatched request 1200002614 for go-e
[ocpp  ] TRACE 2023/03/03 09:21:18 sent JSON message to go-e: [2,"1200002614","RemoteStopTransaction",{"transactionId":0}]
[ocpp  ] TRACE 2023/03/03 09:21:18 started timeout timer for go-e
[ocpp  ] TRACE 2023/03/03 09:21:18 received JSON message from go-e: [3,"1200002614",{"status":"Rejected"}]
[ocpp  ] TRACE 2023/03/03 09:21:18 handling incoming CALL RESULT [1200002614] from go-e
[ocpp  ] TRACE 2023/03/03 09:21:18 completed request 1200002614 for go-e
[lp-1  ] ERROR 2023/03/03 09:21:18 charger: Rejected
[ocpp  ] TRACE 2023/03/03 09:21:18 go-e ready to transmit again
[ocpp  ] TRACE 2023/03/03 09:21:18 timeout canceled for go-e
```

The log shows that evcc tries to stop charging but the go-e just says Rejected.

dajakus, 2023-04-24:
@bkonar6 I see that you were able to change chargeBoxId to petroupoli2 for go-e charger. I'm having trouble connecting go-e charger to steve given that the steve sees connection attempt but the chargeboxID  for go-e charger is empty string "" and I can't add charger to list of charging points if it doesn't have name. How did you managed to change chargeboxID for go-e charger? Is there an option in mobile phone app to configure chargeboxID for go-e charger?

0xFEEDC0DE64, 2023-04-26:
Maybe the chargeboxID should be appended at the end of the url?

EnvironCMO, 2024-10-09:
I send this message:
[
  2,
  "789123",
  "SetChargingProfile",
  {
    "connectorId": 1,  
    "csChargingProfiles": {
      "chargingProfileId": 1,
      "stackLevel": 1,  
      "chargingProfilePurpose": "TxProfile",  
      "chargingProfileKind": "Absolute",  
      "chargingSchedule": {
        "duration": 600,  
        "startSchedule": "2024-10-09T13:50:00Z",  
        "chargingRateUnit": "W",  
        "chargingSchedulePeriod": [
          {
            "startPeriod": 0,
            "limit": 2000
          },
          {
            "startPeriod": 300, 
            "limit": 11000  
          }
        ]
      }
    }
  }
]

and got an accepted back, but it has no effekt