goecharger/go-eCharger-API-v2 #107: Modbus API 2.0 - missing important Parameters

tops4u, 2022-12-10:
Modbus API is still missing crucial Parameters. 

Most importantly is the ability to change number of Phases to be used. Second it is not possible to set a target Charge Amount. So Users still have to stick to HTTP oder MQTT if one of those are to be used on FW 054.9. Please ADD. 

Also the Description of the Modbus Parameters is cumbersome. Why not writing the Register Adresses in the full Format (adding 30001 for READ and 40001 for WRITE)?

0xFEEDC0DE64, 2022-12-10:
where does this 30001 offset come from? Wasnt the offset different for holding registers and input registers? As we support both (I treat modbus function 3 and 4 as the same), how should we write the numbers?

| Register wire-format           | Register as Holding Register | Register as Input Register | Bezeichnung                 | Register Typ      | Datentyp              | Länge | Beschreibung                                                                                                                                                                                                                                                                                                                                               |
| -------------------------------------- | - | - | --------------------------- | ----------------- | --------------------- | ----- | ---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- |
| 100                                    | 40101 | 30101 | CAR_STATE                   | Input Register    | unsigned integer (16) | 1     | **Status PWM Signalisierung**<br>0: unbekannt, Ladestation defekt<br>1: Ladestation bereit, kein Fahrzeug<br>2: Fahrzeug lädt<br>3: Warte auf Fahrzeug<br>4: Ladung beendet, Fahrzeug noch<br>verbunden                                                                                                                                                    |

Would this new table format be ok?

tops4u, 2022-12-11:
> where does this 30001 offset come from? Wasnt the offset different for holding registers and input registers? As we support both (I treat modbus function 3 and 4 as the same), how should we write the numbers?
> Would this new table format be ok?

Yes, I think this would make it more clear for Non-Professional ModBus Users. I have used Modbus before (for Power Converters, but I have had to study the Link in the Top of the Readme). 

Please add the requested Registers as well, otherwise ModBus is not really usable for me.

HeikoG67, 2023-02-17:
Is there an equivalent modbus parameter for API  'frc'?

0xFEEDC0DE64, 2023-02-22:
Did you see my latest change to the modbus spec? FORCE_STATE was added and will be released with the next firmware release

HeikoG67, 2023-02-23:
Thanks for implemeting FORCE_STATE. Just wondering if Register should be 30338 (337) instead 30336 (337) like the latest documentation states.

0xFEEDC0DE64, 2023-02-24:
You were right, this was a copy&paste error. I will fix this soon.

For the go-eController, I am working on a program that automatically counts up the register numbers in our docs (like it does in the firmware implementation). There shouldnt be any human error.

https://raw.githubusercontent.com/goecharger/go-eController-API/main/modbus-de.md
(search for `