goecharger/go-eCharger-API-v2 #108: Auto-Discovery von go-eChargern

mincequi, 2022-12-17:
Hallo zusammen,

gibt es eine Möglichkeit, die go-eCharger im Netzwerk zu _auto-discovern_?
Die offizielle App findet ja alle Wallboxen automatisch. Wie wurde das umgesetzt? Gibt es dafür eine Dokumentation?
Muss man auf eine bestimmte Multicast Adresse hören?

Vielen Dank im voraus!

0xFEEDC0DE64, 2022-12-17:
aktuell announcen die V3 und V4 charger ihren namen über mdns, allerdings sehe ich das nicht als offizielles, dokumentiertes "feature" und möchte mir das recht vorbehalten, den announcten service in zukunft zu ändern, _http._tcp ist etwas blöd gewählt und es gibt immer kollisionen mit druckern, etc. und die app muss danach manuell noch weiter raus filtern.

0xFEEDC0DE64, 2023-03-09:
_go-e_go-eCharger._tcp
_go-e_go-eCharger_V4._tcp
_go-e_controller._tcp

sind die services, die von go-e produkten veröffentlicht werden

closed 2023-03-09 by 0xFEEDC0DE64