goecharger/go-eCharger-API-v2 #109: MQTT(s) mit Prefix Problem (+ api_key, special chars im PW)

stefbos, 2022-12-18:
Zuerst einmal ein dickes Lob. Hatte überlegt mir eine OpenWB zu holen (wegen MQTT, Api, Überschussladen etc.) und mich schlussendlich für die Go e entschieden und bin sehr zufrieden. Insbesondere mit den Möglichkeiten die sie mir als Entwickler eröffnet.

Das einzige was mich wurmt ist MQTT (mit TLS). Kann es sein, dass da im  letzten Update etwas schief gelaufen ist (bin auf 054.9 Beta)?

- Keinerlei Special Chars im Password möglich. Ich nutze gerne Passphrases und trenne die Wörter mit _. Ich konnte einfach keine Verbindung erstellen....erst als ich einen Nutzer mit nur Buchstaben (im Username und Password) erstellt habe ging meine mqtts verbindung. Danke an : https://github.com/goecharger/go-eCharger-API-v2/issues/73.

 - Die WB sendet keine Daten über MQTT(S) sobald ich über die App eine Prefix setze. Lasse ich das Feld leer, werden die Daten auf die oberste Ebene meines Brokers geschickt.

- der api_key tpa liefert immer null. Ein anderer Nutzer hatte das auch schon gemeldet. 
https://github.com/goecharger/go-eCharger-API-v2/issues/96

ratio91, 2023-04-05:
> * Die WB sendet keine Daten über MQTT(S) sobald ich über die App eine Prefix setze. Lasse ich das Feld leer, werden die Daten auf die oberste Ebene meines Brokers geschickt.

Kann ich bestätigen!

logs of my mosquitto, if I set any prefix in the app:
```
 New client connected from 192.168.x.xxx:57374 as go-echarger_213948 (p2, c1, k120, u'admin').
0: Invalid subscription string from 192.168.x.xxx, disconnecting.
0: Client go-echarger_213948 disconnected due to malformed packet.
```

when I set `mtp` to `null` OR if I add a trailing slash (e.g. `go-e/`) it works!

lubbyhst, 2024-03-01:
Thanks for this bug report. This is still an existing problem in v0.55.8.