goecharger/go-eCharger-API-v2 #113: granular charge power control down to mA steps with supported chargers (labeled by e.g. smartWB als OLC)

iStehle, 2023-01-04:
Ich würde ja gerne wenn wieder mehr PV Strom da ist passender Laden können. evcc könnte ja schon alles dafür. Kann bei mir die v3 Hardware das technisch nicht oder ist es einfach noch nicht im Scope das ist das einbaut. 

Ich habe keinen Feature Request dazu gesehen.

chk2902, 2023-01-08:
https://github.com/goecharger/go-eCharger-API-v2/issues/92

0xFEEDC0DE64, 2023-01-09:
steht auf meiner todo liste, allerdings braucht der Change sehr viele testing-ressourcen unserer eigenen pv-regelschleife, diese geht aktuell immer noch von ganzzahligen sprüngen aus und die wieder so zu perfektionieren dauert.

fu-zhou, 2023-01-09:
Was macht denn die "eigene pv-regelschleife"? Gibt es ein API-key, über den man den momentanen PV-Überschuss in Watt oder kW an den Charger übergeben kann?
Der Charger ermittelt aus der Info dann den entsprechenden Strom und Anzahl Phasen inkl. einer Hysterese/ Dämpfung, so dass eine Wolke über der PV Anlage nicht gleich zu riesen Sprüngen in der Ladeleistung führt, was das eine oder andere E-Auto bzw. die verbauten Batterien oder Elektronik nicht verträgt? Damit würde ja ggf. die evcc Fuktionalität im Charger abgebildet...

0xFEEDC0DE64, 2023-01-09:
es gibt noch keinen api key zum Bespielen mit den pv Leistungswerten,
allerdings wird der kommende go-eController (unser smartmeter) selbstständig den Charger finden und pairen, und wenn die Kategorien richtig zu den Mess-Probes zugewiesen wurden, kann sofort mit pPv, pGrid und pAkku überschuss geladen werden.

Hysterese mit Smoothing, automatischer Ladestop/Start, minimaler Ladestrom, automatisches simuluate-unplugging um die 12V Batterie zu entlaseten, Rundungsmodus (PreferFromGrid, PreferToGrid, NormalRounding), automatische Phasenumschlatung, spezielle Kompatilitätseinstellungen für viele Auto-Hersteller, das Zusammenspiel mit dem kommenden dynamischen und statischen Lastmanagement sind die Kern-Features der Regelschleife.

Genaueres kommt noch dazu, bitte noch um etwas Geduld, es wird vermutlich eine api-Schnittstelle geben

fu-zhou, 2023-01-09:
Knaller! Ich bin gespannt...

marcelgsteiger, 2023-02-08:
Vielleicht bin ich eine Spassbremse, aber ich denke, man sollte der ESP32 Firmware nicht zuviel aufbürden. Spätestens wenn mehrere Ladestationen involviert sind, eine PV-Anlage tariflich korrekt einbezogen werden soll oder die Ladevorgänge unterschiedlichen Parteien abgerechnet werden, braucht es ohnehin ein von den Wallboxen unabhängiges lokales System für Steuerung/Messung/Protokollierung/Abrechnung. Dieses kann die "Regelschleife" nebenher auch gleich implementieren und für eine ordentliche Überwachung und Protokollierung sorgen.
Ich nutze hierzu unter anderem NodeRed und API v2. Noch vor der PV-Strommodulation implementierte ich die globale Umschaltung aller go-e Wallboxen auf einphasiges Laden bei geringem Solarertrag oder hohem Stromtarif und das Pausieren einzelner Ladevorgänge zur Glättung der Gesamtlast (z.B. wenn der Elektroboiler der zugehörigen Wohnung im Mehrfamilienhaus eingeschaltet wird). Das lokale System stellt sicher, dass alle Abfrage-, Umschalt- oder Änderungsvorgänge in geeigneten Zeitabständen sequenziell und autokompatibel auf die Ladestationen verteilt werden.

Micha982, 2023-04-17:
Hallo,
gibt es hierzu schon Fortschritte?

_es gibt noch keinen api key zum Bespielen mit den pv Leistungswerten,_ Ist das noch in Zukunft angedacht? Ich würde gerne vermeiden, von jedem Hersteller ein SmartMeter im Schaltschrank zu verbauen ;-)

0xFEEDC0DE64, 2023-04-21:
api key "ids", siehe meine kurzen Versuche den genauer zu erklären:

https://github.com/goecharger/go-eCharger-API-v2/discussions/110#discussioncomment-4973967
https://github.com/goecharger/go-eCharger-API-v2/discussions/110#discussioncomment-5532376
https://github.com/goecharger/go-eCharger-API-v2/discussions/110#discussioncomment-5592204