goecharger/go-eCharger-API-v2 #117: frc setzt sich selbst zurück auf 0

4homeassistant, 2023-01-31:
Ich versuche seit vier Monaten, den Charger dauerhaft zu sperren und laut meiner Recherche und dem dem Support-Kollegen ist frc hier richtig. Wenn ich frc auf 1 setze wird der Wert irgendwann im Laufe eines Tages wieder auf 0 gesetzt ohne Interaktion und ohne dass ein Muster zu erkennen wäre. Beide Charger laufen mit der aktuellen Firmware und Hardware V3.
Gibt's hier die Möglichkeit eines Bugs bzw. kann ich am Charger ein Protokoll auslesen?

0xFEEDC0DE64, 2023-02-01:
Beim Auto abstecken sollte force wieder auf neutral gehen.

wie wärs mit acs immer auf 1 gesetzt lassen (oder über die app die authentifizierung erforderlich machen), und wenn man die Ladung erlauben möchte, dann trx=0 setzen, dann darf man solange laden, bis wieder das auto abgesteckt wurde oder eine minute vergangen ist ohne dass ein auto dran hängt?

der force state verwenden die leute eher für unterbrechungen während dem laden.

4homeassistant, 2023-02-01:
Danke @0xFEEDC0DE64 für die schnelle Antwort. Ich steuere die Charger über Home Assistant und mein Ziel ist, Laden grundsätzlich nur mit Authentifizierung und zu eingestellten Zeiten zu ermöglichen.
Dazu ist `acs` und `frc` immer auf `1`. `frc` wird nur nach einer fest eingestellten Zeit auf `0` gesetzt und wieder auf `1` sobald die Endzeit erreicht ist. Laden ist also nur innerhalb eines gesetzten Zeitfensters und nach Authentifizierung möglich.

Aus welchem Anlass wird `frc` denn auf `0` gesetzt wenn das Fahrzeug abgesteckt wird? Das Problem ist, dass sich `frc` scheinbar auch zu zufälligen Zeiten auf `0` setzt. Also auch wenn kein Fahrzeug verbunden ist. Gibt es eine andere Option außer `frc` um den Charger zu sperren und reagiert `frc` auch auf andere Events? Ist irgendeine Art Logging mit den Chargern möglich?

4homeassistant, 2023-02-01:
Danke Dir @cbrandlehner. Ja, die Home Assistant Integration, die ich nutze, setzt auch auf MQTT. Da wäre es sicher möglich, über die `file` Plattform ein CSV-Format zu definieren und den Output dort reinzuschreiben. Ich vermute aber, dass das nicht den Grund anzeigt, weshalb der betreffende Wert geändert wird, sondern nur, dass er geändert wurde.

cbrandlehner, 2023-02-01:
@4homeassistant gut möglich. Eventuell kann man ein Muster daraus erkennen und dadurch auf eine Ursache schließen. 🤷

flod-1, 2023-02-02:
Ich habe dasselbe Verhalten bemerkt. Ich würde fast gar sagen, dass es immer beim Ausstecken des Autos zurück auf 0 springt. So zumindest sieht es auch in der Grafik in der App aus, die für die letzten stunden zeigt, ob Laden erlaub war, etc.
Aus meiner Sicht wirklich sehr nervig, wenn das Laden durch ein SmartHome gesteuert werden soll, ja. Ich hoffe daher auf baldige Korrektur :)

Als Workaround gehe ich versuchsweise im SmartHome hin und setze frc nochmals auf 1, wenn erkannt wird, dass a) kein Auto (mehr) verbunden ist, b) frc auf 0 springt, c) die SmartHome Logik das Laden zu dem Zeitpunkt noch immer unterbindet möchte. Ob der Workaround stabil funktioniert ist noch nicht klar ... er ist jedenfalls sehr unschön ;)

4homeassistant, 2023-02-03:
Hi @flod-1, ja - beim Ausstecken und Neustarten ist das immer der Fall. Allerdings ist es bei unseren Chargern auch so, dass sich `frc` auch ohne Ausstecken und willentlich ausgelöstem Neustart wieder auf `0` setzt.
@0xFEEDC0DE64, kannst Du mir denn sagen, weshalb das so ist? Vielleicht hilft es dabei, die Intention hinter dieser Funktion besser zu verstehen. Wenn ich zwischen den Zeilen lese, sollte `frc` aber nur in diesem Fällen auf `0` gesetzt werden. In welchen Fällen würde dieses Verhalten denn noch greifen? Ohne Logging ist es schwierig, dahinterzukommen. Es fühlt sich in jedem Fall nach einem Bug an.

flod-1, 2023-02-03:
Aus meinem Verständnis heraus ist es definitiv auch ein Bug, da frc laut Beschreibung doch genau für das Erzwingen von Laden an/aus genutzt werden soll. Und genau das Erzwingen geht hier nicht zuverlässig.

4homeassistant, 2023-03-27:
Für den Moment habe ich mich auch mit einer Logik mit einer Home Assistant Automation beholfen, die bei Änderung von ```frc``` den Status von ```car``` prüft und ggf. ändert/korrigiert.

Da die Doku das beschriebene Verhalten nicht wiedergibt und es logisch nicht zu erklären ist bleibt's auch in meinen Augen ein Fehler.
@0xFEEDC0DE64 Daniel, vielleicht könnt Ihr die Doku dahingehend ergänzen oder das Verhalten nachvollziehbarer machen. Kannst Du uns denn einen Grund nennen weshalb ```frc``` beim Abstecken automatisch auf 0 geht? Bei Neustart durch manuelle Auslösung oder kurze Stromlosigkeit auch.

gyroslav, 2023-04-21:
Habe das gleiche Verhalten. Ich schalte FRC per Skript auf 1 wenn CAR=1. Damit der Ladevorgang beim nächsten Anstecken nicht sofort losgeht, und auf genug PV geprüft wird.

skreibig7, 2024-04-03:
Hallo in die Runde,

ich bin relativ neu im Bereich Home Assistant unterwegs. Stehe gerade vor der gleichen Herausforderung: Möchte, dass eine Ladung nur dann beginnt, wenn ich es möchte: Beispiel PV- Überschuss- Laden. Könnt ihr mir sagen, wir ihr es gelöst habt, FRC auf 1 oder 0 zu setzen? Ich würde gerne das Auto anstecken, und dann via Switch (Oder Automation) sagen, dass es jetzt laden kann. 

Vielen Dank :)

4homeassistant, 2024-04-03:
@skreibig7, das hier ist das go-e Charger API v2 Repository und in diesem Issue geht es um einen möglichen Bug und dessen Auswirkung an der API. Du bist hier generell mit Fragen zu Funktionsweisen und im Speziellen mit Fragen zu Home Assistant komplett falsch. Fragen zur Funktionsweise der API kannst Du hier in den Repos unter "Discussions" stellen, Fragen zu Home Assistant in der HA-Community: https://community.home-assistant.io
Hier gibt's auch einen Thread von Sebastian Muszynski aka syssi, der die go-e Charger HA-Integration geschrieben hat:
https://community.home-assistant.io/t/custom-component-goecharger-mqtt-go-echarger-homefix-ev-charger-integration/394420/20