goecharger/go-eCharger-API-v2 #118: RFID Karte bei API mitgeben

philg1234, 2023-02-09:
Hey,

ist es möglich per API v2 wenn die GO-E auf Authentifzierung steht die RFID Karte mitzugeben und somit die Authentifzierung freizuschalten und danach den Ladevorgang zu starten?

Wenn ich per API die Authentifzierung auf Open setzte und ein Auto an der Ladestation hängt, fängt es sofort an zu laden.
Es wird aber natürlich alles auf die RFID 0 getaggt.

Gerne würde ich auch darüber alles auf einen bestimmten RFID Chip taggen.

Grüße
Phil

0xFEEDC0DE64, 2023-02-09:
trx=1 sollte der erste rfid karten "slot" sein, zum energie drauf buchen.
trx=0 erlaubt das laden, ohne dass die energie wo drauf gebucht wird.
mit der neuesten firmware kann man auch per http api die hingehaltene rfid-seriennummer auslesen, entweder die seriennummer vom zuletzt hingehaltenen oder die seriennummer, mit dem die transaktion damals gestartet wurde.

philg1234, 2023-02-10:
Danke für deine Antwort.

Heißt also es ist momentan nicht möglich vor dem Ladevorgang ein RFID Chip mitzugeben und diesen dann zu starten?
Die Möglichkeit mit dem trx=0 werde ich gleich einmal testen.

Grüße

4homeassistant, 2023-02-12:
@philg1234, lade Dir die aktuelle Beta-Firmware 055.2 auf Deinen Charger. Damit funktionieren mit `trx` auch Werte > `0`. Ich hatte dazu diesen Thread eröffnet: https://github.com/goecharger/go-eCharger-API-v2/issues/116
Wenn das für Dich passt kannst Du diesen Thread hier gerne schließen. :-)

closed 2023-02-13 by philg1234