goecharger/go-eCharger-API-v2 #120: Fahrzeugerkennung

herbert131, 2023-03-02:
Hi,
nach Ladeende bleibt die WB auf car=4 (complete). Wird das Kabel a**m Fahrzeug** entfernt ändert sich der Status nicht. Wie erkenne ich ein ausgestecktes Ladekabel am Fahrzeug nach dem Laden ?
Danke

0xFEEDC0DE64, 2023-03-03:
eigentlich sollte der status auf Idle zurückgehen, ist es eine V3 oder V4?

0xFEEDC0DE64, 2023-03-28:
kann ich so nicht nachstellen, falls das noch ein problem ist, bitte mit der seriennummer beim support melden.

mleck28, 2024-04-05:
Ich habe das gleiche Problem. In Node-Red per mqtt bleibt das Auto auf "4" stehen, in der App steht, dass kein Auto verbunden ist. 
Gibt es hierzu schon etwas neues?

[image]

[image]