goecharger/go-eCharger-API-v2 #122: Cloud Api Request 404 --> "Data is outdated"

stefbos, 2023-03-11:
Hey, habe seit heute Mittag Probleme mit der Cloud Api. Wenn ich Requests Lokal ausführe klappen Sie wunderbar, aber über die Cloud Api bekomme ich:

`
{
  "success": false,
  "reason": "Data is outdated",
  "age": 25382.09100008011
}
`
Das Token ist valid, wenn ich ein anderes schicke bekomme ich nen 401er.

Egal bei welchem Request..ich kriege immer den 404. Ich habe nichts an meinem Code geändert (Steuere die Wallbox und Überschussladen über ne eigene Node.js App).Hat auch immer bestens geklappt. Ich konnte zu dieser Fehlermeldung auch nichts finden (Google, Issues). Auch ein Update der Firmware auf die neuste Beta hat nicht geholfen.

Über Hilfe wäre ich sehr dankbar.

0xFEEDC0DE64, 2023-03-13:
Das Problem liegt an unserem Cloudserver, der ist leider seit Wochen hoffnungslos überlastet und nicht darauf vorbereitet, automatisch auf mehrere Hosts zu skalieren. Wir holen das gerade nach, allerdings wird es sicher noch die Woche dauern, bis der in Betrieb geht. Ich schreib gern nochmal, sobald ein zweiter Server in Betrieb geht.

0xFEEDC0DE64, 2023-03-28:
Sollte inzwischen längst gefixt sein